carlosmccosta/dynamic_robot_localization #1: Issues installing on Kinetic

cristian-frincu, 2017-03-16:
Running On Ubuntu 15.10 Linux version: 4.2.0-42-generic

1 ) I have cloned the repo in ~/
2) Ran bash install.sh

There are a couple warnings about deprecated functions, than the error shown bellow:
[image]

Can you please help me with the install? 

Thank you

carlosmccosta, 2017-03-16:
Good night :)

You need to compile pcl from my fork.
Those missing files were developed by me and are not yet merged into the pcl official repository.

Please read the notes on the readme.md:
The kinetic-devel branch requires to compile PCL from source using branch master-all-pr from https://github.com/carlosmccosta/pcl (you can use this package.xml (https://gist.github.com/carlosmccosta/1ec3e3bdce419441b0b17bf9bb707552) to compile it within a catkin tools workspace).
To ensure that the same pcl libraries are used in drl and its dependencies, the following packages should also be compiled:
pcl_conversions
pcl_msgs
perception_pcl

Let me know if you need further help :)
Have a nice day :)

cristian-frincu, 2017-03-18:
Hello Carlos,
Thank you for the quick reply.
I have followed you instruction, and I have compiled pcl from your repo. The error went away.

I am now getting a different problem, when I am trying to run the localization launch file:
[image]

I am running it by being navigating to ~/catkin_ws/src and running "roslaunch dynamic_robot_localization/launch/dynamic_robot_localization_system.launch"

Here is a structure of my catkin_ws:
[image]

Any suggestion what might be causing this error?

Thank you for your time :)

carlosmccosta, 2017-03-18:
Good night :)

This type of error happens when ros can't locate the packages in the launch file.
Check that you are sourcing the catkin_ws/devel/setup.bash
http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment
If you are, then
run the command
rospack profile
And then check if you can find the main package
rospack find dynamic_robot_localization

Have a nice day :)

cristian-frincu, 2017-03-24:
Hello,
I have sourced the source.bash,
The rospack can find the package.
The errors are still there. 
The only part where I deviated from the install, is that I used catkin_make_independent. This is because PCL was a plain make package.

Any suggestion?

Have a nice day :)

carlosmccosta, 2017-03-24:
Good night :)

The errors shown happen when a given binary can't be found inside a package.
Maybe the mix of catkin_make and catkin_make_isolated might be causing problems.
I am using the improved catkin build system named catkin-tools, which can compile cmake packages directly.

Try migrating your workspace from catkin to catkin-tools:
http://catkin-tools.readthedocs.io/en/latest/migration.html
http://catkin-tools.readthedocs.io/en/latest/quick_start.html

I also compiled pcl in a workspace and the remaining of the packages in another workspace (overlay) :
http://catkin-tools.readthedocs.io/en/latest/mechanics.html#workspace-chaining-extending
http://wiki.ros.org/catkin/Tutorials/workspace_overlaying

Have a nice day :)

carlosmccosta, 2018-09-17:
Good afternoon :)

For future reference, I updated the install scripts for compiling this package with either the older [catkin_make_isolated](http://www.ros.org/reps/rep-0134.html) or the newer [catkin_tools](https://catkin-tools.readthedocs.io/en/latest/migration.html).
Also, during the testing of the scripts I notice that when catkin_make_isolated is used, the setup.bash that needs to be sourced is different.
Namely, it is not devel/setup.bash but **devel_isolated/setup.bash**.
I never had this problem because I have been using catkin_tools (which supports the build of pure CMake packages).
As such, I never needed to use catkin_make_isolated to compile PCL.

I hope that the [updated install scripts](https://github.com/carlosmccosta/dynamic_robot_localization/tree/kinetic-devel/install) and [readme](https://github.com/carlosmccosta/dynamic_robot_localization#installation-and-package-dependencies) will provide an easy way to install and use this package.

Have a nice day :)

closed 2018-09-22 by carlosmccosta