carlosmccosta/dynamic_robot_localization #2: How to start

melhousni, 2018-11-08:
Hello
Im trying to use this package but it's a little bit overwhelming!
Let's suppose I have a rosbag with pointcloud and imu data, what would be the first steps to generate a map and use it for navigation? Thank you

carlosmccosta, 2018-11-08:
Good night :)

The 3D processing pipeline can be configured for different use cases, such as 3/6 dof mapping or 3/6 dof localization of a robot / sensor / object.
Have a look at the different launch files (and their associated yaml files that specify the pipeline configuration) for each use case (in [dynamic_robot_localization_tests](https://github.com/carlosmccosta/dynamic_robot_localization_tests/tree/hydro-devel/launch/environments) and below) and let me know if you need any help :)

If you need fast pose estimation (such as +- 20 or 30 Hz), I would suggest to first make the map without setting [registration time limits](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L539) and playing the bag file at low rate. This way you can generate a high quality map with [surface estimation](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L344) using the [moving least squares](http://pointclouds.org/documentation/tutorials/resampling.php).
Then you can tune the processing pipeline [ICP registration parameters](http://docs.pointclouds.org/trunk/classpcl_1_1_iterative_closest_point.html) according to your requirements of registration accuracy and processing time.
For example, for higher accuracy, you can set the convergence thresholds ([transformation_epsilon](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L458) | [euclidean_fitness_epsilon](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L459) | [convergence_absolute_mse_threshold](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L536) | [convergence_rotation_threshold](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L537)) with low values and let the ICP max iterations run higher ([max_number_of_registration_iterations](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L460)) while also setting the [convergence_time_limit_seconds](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/yaml/schema/drl_configs.yaml#L509) also higher.

Example of launch files and videos for each use case:
- 6 DoF localization / mapping of a flying kinect (using the [ethzasl iros dataset](https://projects.asl.ethz.ch/datasets/doku.php?id=kinect:iros2011kinect))
    * [Launch file](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/asl/ethzasl_kinect_dataset.launch)
    * [Video](https://www.youtube.com/watch?v=Us9XkcRx_5I)
- 6 DoF robot localization (map from CAD data)
    * [Launch file](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/labiomep/labiomep_6dof.launch)
    * [Robot localization video](https://www.youtube.com/watch?v=VtOv4beLcBM)
- 6 DoF object pose estimation
    * [Launch file](https://github.com/carlosmccosta/assembly_projection_mapping_teaching/blob/kinetic-devel/launch/object_recognition.launch)
    * [Pose estimation video with projection mapping](https://www.youtube.com/watch?v=557vglPW6Ko)
    * [Pick and place](https://www.youtube.com/watch?v=ic9rhJEr_EE)
- 3 DoF robot localization / mapping
    * [Launch file](https://github.com/inesc-tec-robotics/guardian_config/blob/hydro-devel/localization/guardian_localization.launch)
    * [Robot mapping video](https://www.youtube.com/watch?v=GKDG7c61OLo)
    * [Robot localization video](https://www.youtube.com/watch?v=CIP-QH87z8E)

Have a nice day :)

kkVishnu, 2021-03-11:
@carlosmccosta ,
I am really new to this dynamic robot localization pipeline processing.

As a beginner, I am trying to launch the file "ethzasl_kinect_dataset.launch" using the ethzasl dataset. So when i check the package 'dynamic_robot_localization_tests' I found some use full launch files like ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch in /dynamic_robot_localization_tests/launch/environments/asl/bags/ . 

So, i want to see the same output referred in the video https://www.youtube.com/watch?v=UslDiUkm7wE (Video 6: Free fly test with Kinect in the ETHZ RGB-D dataset using the 6 DoF localization system) in my Ubuntu Laptop.
For that I am doing following steps: 
> source /home/catkin_ws_drl/devel_release/setup.bash
> cd /home/catkin_ws_drl/dynamic_robot_localization_tests/launch/environments/asl/bags/
> roslaunch ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch

--> After this step Rviz got displayed and then I pressed space bar from the current terminal. Now point clouds are displaying but there are some mismatches in my output when i compare it to the video https://www.youtube.com/watch?v=UslDiUkm7wE .

The output I received is recorded as a video (Record1.mkv) and added at https://github.com/kkVishnu/DRLVideo.git
It would be really helpful if you can confirm whether any additional steps needs to be done at my end ? or Is there any modification expected in ethzasl_kinect_dataset.launch or ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch to get the same output?

Development environment : 
OS : Ubuntu 18.04
ROS: ROS Melodic
RAM: 16GB
Processor: Intel® Core™ i7-10810U CPU @ 1.10GHz × 12 
Graphics: Intel® UHD Graphics (CML GT2)

carlosmccosta, 2021-03-15:
Hello,

Thank you for reporting the problem :)

Please check the response I gave below, in which I discussed this problem and its solution.
https://github.com/carlosmccosta/dynamic_robot_localization/issues/27

Have a nice day,

kkVishnu, 2021-03-16:
Hi @carlosmccosta ,
Thank you for the detailed information and now I am getting the same output.

I have one doubt, when i checked the rostopic list i found multiple topics for pose like /dynamic_robot_localization/localization_pose , /dynamic_robot_localization/localization_poses etc . If i understand correctly the topic "**/dynamic_robot_localization/localization_pose**" is used for visualizing 6DoF sensor pose in Rviz viewer.
Could you please confirm the exact topic name to get the 6DoF sensor pose ?

https://user-images.githubusercontent.com/80447835/111301403-2f23b980-8678-11eb-9bdd-cc837796ba2e.mp4

carlosmccosta, 2021-03-16:
Hello,

I updated the rviz file for having labels that are easier to understand (please do git pull in the `dynamic_robot_localization_tests`).
- `/dynamic_robot_localization/localization_pose [geometry_msgs/PoseStamped]` -> 6 DoF pose estimated by drl for the last point cloud
- `/dynamic_robot_localization/localization_poses [geometry_msgs/PoseArray]` -> Array with all the (red) poses estimated by drl
- `/dynamic_robot_localization/groundtruth_poses [geometry_msgs/PoseArray]` -> Array with the (green) ground truth poses associated with each drl pose (for every drl pose, I lookup in TF [the associated ground truth pose](https://github.com/carlosmccosta/robot_localization_tools)) -> the ground truth was generated by Vicon cameras (for more info, check the [dataset page](https://projects.asl.ethz.ch/datasets/doku.php?id=kinect:iros2011kinect))
- `/trajectory [nav_msgs/Path]` -> Red path associated with the drl poses
- `/trajectory_gt [nav_msgs/Path]`-> Green path associated with the ground truth poses

[image]

Have a nice day,

kkVishnu, 2021-03-18:
Hi @carlosmccosta ,

Thanks for the updated info.

I am trying to check the covariance data in 6DoF pose , so I used below command to check the  covariance matrix after launching ethzasl_kinect_dataset (roslaunch dynamic_robot_localization_tests ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch) :
$ rostopic echo /dynamic_robot_localization/localization_pose_with_covariance

And I am getting zero values in covariance matrix like below:
-----------------------------------------------
header: 
  seq: 0
  stamp: 
    secs: 1298112282
    nsecs: 131209000
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 1.33109009299
      y: -0.721230991265
      z: 1.64216395765
    orientation: 
      x: 0.034314302265
      y: 0.27709929409
      z: -0.086945950686
      w: 0.956283907391
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 1
  stamp: 
    secs: 1298112282
    nsecs: 163552000
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 1.33042057397
      y: -0.719728996275
      z: 1.64075392505
    orientation: 
      x: 0.0371451415499
      y: 0.27721823527
      z: -0.0889289796376
      w: 0.955961257099
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 2
  stamp: 
    secs: 1298112282
    nsecs: 197002000
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 1.33172996881
      y: -0.724027524406
      z: 1.63992342201
    orientation: 
      x: 0.0387589196238
      y: 0.278446604681
      z: -0.0881394768412
      w: 0.955613241386
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
.
.
---
header: 
  seq: 492
  stamp: 
    secs: 1298112298
    nsecs: 713698000
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 1.51738627546
      y: -0.422942714822
      z: 1.62615816055
    orientation: 
      x: -0.030203352863
      y: 0.309130183314
      z: 0.126192669006
      w: 0.942126157969
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 493
  stamp: 
    secs: 1298112298
    nsecs: 745706000
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 1.51747147745
      y: -0.424794380711
      z: 1.61823682282
    orientation: 
      x: -0.030589948911
      y: 0.295414033612
      z: 0.142660588958
      w: 0.944162464901
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
-----------------------------------------------

So, Is there any modification required in config files to see the non zero values in covariance matrix ?

carlosmccosta, 2021-03-22:
Hello,

For enabling [covariance estimation](https://github.com/inesc-tec-robotics/dynamic_robot_localization/blob/noetic-devel/yaml/schema/drl_configs.yaml#L759) only within drl [you can set](https://github.com/inesc-tec-robotics/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L28):
```
<arg name="use_ukf_filtering" default="false" />
<arg name="yaml_configuration_covariance_filename" default="$(find dynamic_robot_localization)/yaml/configs/covariance_estimation/covariance_estimation.yaml" />
```

For enabling covariance estimation within drl for using the tracking from ukf_localization_node from [robot_localization](http://wiki.ros.org/robot_localization) package, you can enable:
```
<arg name="use_ukf_filtering" default="true" />
```

It is disabled by default because during the tests I made the time it takes to compute the covariance was longer than the entire drl pipeline and it was not resulting in better tracking when paired with the robot_localization. Also, the covariance estimation accuracy was not high.

As such, if you really need the covariance, you can enabled it in drl.
If you only need the 6 DoF tracking, I suggest keeping it disabled.

Have a nice day,