carlosmccosta/dynamic_robot_localization #8: Compile the project once

abylikhsanov, 2019-09-20:
Compiling on ARM takes a lot of time, especially PCL! What is the best way to compile PCL once for all and save its binaries and libraries so I can simply transfer to another ARM machine? I tried to simply make the PCL independently and then install its binaries to some folder and use that but getting an error at:

```
[ 37%] Linking CXX shared library ../lib/libpcl_segmentation.so
collect2: error: ld returned 1 exit status
segmentation/CMakeFiles/pcl_segmentation.dir/build.make:641: recipe for target 'lib/libpcl_segmentation.so.1.8.1.99' failed
```

carlosmccosta, 2019-09-20:
Good morning :)

You can use catkin to create an "installable" version of the packages.

For enabling it in a catkin_tools workspace:
`catkin config --install`

For disabling it:
`catkin config --no-install`

More info:
[https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_config.html#installing-packages](https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_config.html#installing-packages)

The older catkin_make can also generate the install space:

[http://wiki.ros.org/catkin/Tutorials/using_a_workspace#Installing_Packages](http://wiki.ros.org/catkin/Tutorials/using_a_workspace#Installing_Packages)

You can also generate a deb package from the install space for making it easier to install.
There are several ways to achieve this:
* [bloom](http://wiki.ros.org/bloom)
  * [https://answers.ros.org/question/173804/generate-deb-from-ros-package/?answer=173845#post-id-173845](https://answers.ros.org/question/173804/generate-deb-from-ros-package/?answer=173845#post-id-173845)
* [CheckInstall](https://wiki.debian.org/CheckInstall)
* [dpkg-buildpackage](http://manpages.ubuntu.com/manpages/precise/man1/dpkg-buildpackage.1.html)
* [CPack](https://cmake.org/cmake/help/v3.0/module/CPack.html)

abylikhsanov, 2019-09-20:
Thanks :)  But the error I am getting, do you have an answer for that? Because even if I create a .deb package of your version PCL I might still get that issue

carlosmccosta, 2019-09-20:
You seem to have in /home/abyl/right-wheel/src/laser_icp a target conflicting with pcl_ros
Both seem to be generating pcl_ros_gencfg.
Check the CMakeLists.txt of laser_icp

abylikhsanov, 2019-09-20:
Yes but I do not see any conflicts in my CMakeLists, may be you can spot something irregular?

```
cmake_minimum_required(VERSION 2.8.3)
project(laser_icp)

add_compile_options(-std=c++17)

find_package(catkin REQUIRED COMPONENTS
  roscpp
  nodelet
  sensor_msgs
  tf2
  tf2_ros
  pcl_ros
  pcl_conversions
  geometry_msgs
  nav_msgs
)

find_package(catkin REQUIRED)
find_package(PCL REQUIRED)
find_package(PkgConfig)

include_directories(include ${catkin_INCLUDE_DIRS} ${PCL_INCLUDE_DIRS})

catkin_package(
  INCLUDE_DIRS include
  LIBRARIES laser_icp
  CATKIN_DEPENDS roscpp
)
```

carlosmccosta, 2019-09-20:
The error is pointing at conflicts with dynamic_reconfigure here:
[https://github.com/ros-perception/perception_pcl/blob/melodic-devel/pcl_ros/CMakeLists.txt#L70](https://github.com/ros-perception/perception_pcl/blob/melodic-devel/pcl_ros/CMakeLists.txt#L70)

Unless you remove parts of the cmakelists above, I do not see conflicts.
I normally use a cmake variable for dealing with package dependencies (that should be included in find_package and catkin_depends).
Look here for an example:
[https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/CMakeLists.txt#L10](https://github.com/carlosmccosta/dynamic_robot_localization/blob/kinetic-devel/CMakeLists.txt#L10)

You also have a repeared find_package of catkin and in catkin_package are declaring a library that was not created with add_library (laser_icp).

abylikhsanov, 2019-09-20:
What parts of the Cmakelists should be removed?

carlosmccosta, 2019-09-20:
Sorry, I meant that it looks like you did not posted the full contents of the cmakelists file.
For example, you declared a library in catkin_package but it was not created with add_library.

abylikhsanov, 2019-09-20:
Oh okay, but I do not think it causes any errors, here is the rest:
```
add_library(laser_icp src/laser_icp.cpp)

# We don't link against pcl because we are using header only parts
target_link_libraries(laser_icp ${catkin_LIBRARIES})
add_dependencies(laser_icp ${catkin_EXPORTED_TARGETS})

# Nodelet
add_library(laser_icp_nodelet src/laser_icp_nodelet.cpp)
target_link_libraries(laser_icp_nodelet laser_icp)

# Node
add_executable(laser_icp_node src/laser_icp_node.cpp)
target_link_libraries(laser_icp_node laser_icp)

```

So I actually used the Cmake variable to set the ROS packages and used find catkin, like in your example and it worked.. Interesting, does it mean I was declaring them twice which caused the error? I am reading about cmake once more... Never encountered such problem before.

carlosmccosta, 2019-09-20:
If you deviate from the normal usage of cmakelists.txt and package.xml (such as finding catkin twice, not exporting deps, ...) you may encounter strange errors such as the one you mentioned.

For making you life as pain free as possible, I suggest you follow the advice given in these docs:
* [http://wiki.ros.org/catkin/CMakeLists.txt](http://wiki.ros.org/catkin/CMakeLists.txt)
* [http://wiki.ros.org/catkin/package.xml](http://wiki.ros.org/catkin/package.xml)

closed 2019-09-20 by abylikhsanov

zoombinis, 2021-02-26:
@abylikhsanov what was the solution? I get this when compiling two projects that both use pcl. If my ROS_WS only contains one project with PCL I do not get this same issue as you.

carlosmccosta, 2021-03-02:
Hello,

I suggest you create a variable for the package dependencies and use it in `find_package` and `catkin_depends`:

Variable:
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/CMakeLists.txt#L10

Places where to use it:
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/CMakeLists.txt#L39
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/CMakeLists.txt#L86

Have a nice day,