carlosmccosta/dynamic_robot_localization #11: drl build linking error

bill4u, 2019-10-14:
I have installed: ros-kinetic-laserscan-to-pointcloud_1.0.0-0xenial_amd64.deb
ros-kinetic-pcl_1.8.1-0xenial_amd64.deb

bill4u, 2019-10-14:
Do I need to remove the system installed PCL and reinstall  ros-kinetic-pcl_1.8.1-0xenial_amd64.deb?

bill4u, 2019-10-14:
solved.THX

carlosmccosta, 2019-10-14:
Good morning :)

I packaged and configured my fork of pcl for overriding the system installation (this way you do not need to touch your installation of pcl).

The error you are getting probably has the same cause of [this one](
https://github.com/carlosmccosta/dynamic_robot_localization/issues/10#issuecomment-538740122).

I integrated the [commit](https://github.com/carlosmccosta/dynamic_robot_localization/commit/1b279877308a34ce4e526ab6e190ba67c1eb4a2e) that solved this issue from the melodic to the kinetic branch.
Please do git pull and see if it solves the linking error.

P.S.
I did not have this issue in Ubuntu 16, probably because I am using an upgraded version of gcc (7.4) while also using [catkin tools](https://catkin-tools.readthedocs.io/en/latest/migration.html).

closed 2022-02-15 by bill4u