carlosmccosta/dynamic_robot_localization #20: build with ros-noetic

sunrentao, 2020-10-23:
Hi Carlos,

Thank you for providing this good package, but here I have a problem when build the package with ros-noetic.
I have install your customized PCL library but I still get this error, do you have any idea?

<pre>[ 30%] <font color="#4E9A06">Building CXX object dynamic_robot_localization/CMakeFiles/drl_common.dir/src/common/registration_visualizer.cpp.o</font>
In file included from <b>/usr/include/boost/detail/endian.hpp:9</b>,
                 from <b>/usr/local/include/pcl-1.8/pcl/PCLPointCloud2.h:11</b>,
                 from <b>/usr/local/include/pcl-1.8/pcl/pcl_base.h:55</b>,
                 from <b>/usr/local/include/pcl-1.8/pcl/registration/registration.h:45</b>,
                 from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/registration_visualizer.h:41</b>,
                 from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:39</b>,
                 from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43</b>:
<b>/usr/include/boost/predef/detail/endian_compat.h:11:161:</b> <font color="#06989A"><b>note: </b></font>#pragma message: The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include &lt;boost/predef/other/endian.h&gt; and use BOOST_ENDIAN_*_BYTE instead
   11 | ORDER is deprecated. Please include &lt;boost/predef/other/endian.h&gt; and use BOOST_ENDIAN_*_BYTE instead&quot;<font color="#06989A"><b>)</b></font>
      |                                                                                                       <font color="#06989A"><b>^</b></font>

In file included from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43</b>:
/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp: In instantiation of ‘<b>bool dynamic_robot_localization::RegistrationVisualizer&lt;PointSource, PointTarget&gt;::setRegistration(pcl::Registration&lt;PointSource, PointTarget&gt;&amp;) [with PointSource = pcl::PointXYZRGBNormal; PointTarget = pcl::PointXYZRGBNormal]</b>’:
<b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:52:1:</b>   required from here
<b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:70:2:</b> <font color="#CC0000"><b>error: </b></font>no matching function for call to ‘<b>pcl::Registration&lt;pcl::PointXYZRGBNormal, pcl::PointXYZRGBNormal, float&gt;::registerVisualizationCallback(std::function&lt;void(const pcl::PointCloud&lt;pcl::PointXYZRGBNormal&gt;&amp;, const std::vector&lt;int&gt;&amp;, const pcl::PointCloud&lt;pcl::PointXYZRGBNormal&gt;&amp;, const std::vector&lt;int&gt;&amp;)&gt;&amp;)</b>’
   70 |  <font color="#CC0000"><b>registration</b></font>.registerVisualizationCallback(this-&gt;update_visualizer_);
      |  <font color="#CC0000"><b>^~~~~~~~~~~~</b></font>
In file included from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/registration_visualizer.h:41</b>,
                 from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:39</b>,
                 from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43</b>:
<b>/usr/local/include/pcl-1.8/pcl/registration/registration.h:380:7:</b> <font color="#06989A"><b>note: </b></font>candidate: ‘<b>template&lt;class FunctionSignature&gt; bool pcl::Registration&lt;PointSource, PointTarget, Scalar&gt;::registerVisualizationCallback(boost::function&lt;FunctionSignature&gt;&amp;) [with FunctionSignature = FunctionSignature; PointSource = pcl::PointXYZRGBNormal; PointTarget = pcl::PointXYZRGBNormal; Scalar = float]</b>’
  380 |       <font color="#06989A"><b>registerVisualizationCallback</b></font> (boost::function&lt;FunctionSignature&gt; &amp;visualizerCallback)
      |       <font color="#06989A"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/usr/local/include/pcl-1.8/pcl/registration/registration.h:380:7:</b> <font color="#06989A"><b>note: </b></font>  template argument deduction/substitution failed:
In file included from <b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43</b>:
<b>/home/rentao/wheel/develop/tiradar/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:70:2:</b> <font color="#06989A"><b>note: </b></font>  ‘<b>std::function&lt;void(const pcl::PointCloud&lt;pcl::PointXYZRGBNormal&gt;&amp;, const std::vector&lt;int&gt;&amp;, const pcl::PointCloud&lt;pcl::PointXYZRGBNormal&gt;&amp;, const std::vector&lt;int&gt;&amp;)&gt;</b>’ is not derived from ‘<b>boost::function&lt;Signature&gt;</b>’
   70 |  <font color="#06989A"><b>registration</b></font>.registerVisualizationCallback(this-&gt;update_visualizer_);
      |  <font color="#06989A"><b>^~~~~~~~~~~~</b></font>
make[2]: *** [dynamic_robot_localization/CMakeFiles/drl_common.dir/build.make:193: dynamic_robot_localization/CMakeFiles/drl_common.dir/src/common/registration_visualizer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7231: dynamic_robot_localization/CMakeFiles/drl_common.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<font color="#CC0000">Invoking </font><font color="#CC0000"><b>&quot;make -j8 -l8&quot;</b></font><font color="#CC0000"> failed</font>
</pre>

carlosmccosta, 2020-10-23:
Hello,

From your log, it seems that cmake is loading the system version of pcl (version 1.8), instead of my fork of pcl (version 1.11).
Please check that you have the last commit of my fork of pcl (do a `git pull` on branch `master-all-pr`):
https://github.com/carlosmccosta/pcl

Then compile my fork of pcl inside your catkin workspace (this way it overrides the system version of pcl) and confirm that when you do `roscd pcl`, it goes to the src folder of my fork of pcl.

I made install scripts to make it easier to install drl:
https://github.com/carlosmccosta/dynamic_robot_localization#installation-from-source
https://github.com/carlosmccosta/dynamic_robot_localization/blob/melodic-devel/install/e_build.bash

You can check if my fork of pcl is being selected by catkin with (check if `PKG_CONFIG_PATH` has my pcl fork folder):
```
catkin build --get-env dynamic_robot_localization
```

For confirming that the .so libraries from my fork of pcl were linked with drl, you can use ldd:
```
ldd $HOME/catkin_ws/devel_release/lib/dynamic_robot_localization/drl_localization_node | grep pcl
```

Have a nice day :)

sunrentao, 2020-10-23:
yes, this works. thank you. Have a nice day :)

closed 2020-10-23 by sunrentao