carlosmccosta/dynamic_robot_localization #21: ROS2 Support

SeungRyeol, 2020-11-11:
Hello, Carlos

I am very surprised by your package performance.

I want to use this package on ROS2 as well.

Is there any way to use it?

Or do you have plan to support ROS2 for this package.

carlosmccosta, 2020-11-11:
Hello,

I am planning the migration to ROS 2 during next year.
Currently a lot of the sensors drivers that I am using are not yet in ROS 2.

But you can communicate between ROS 1 and ROS 2 nodes using the ros1_bridge:
- https://github.com/ros2/ros1_bridge
- https://www.theconstructsim.com/how-to-communicate-between-ros1-ros2-with-ros1_bridge

Have a nice day :)

SeungRyeol, 2020-11-12:
Thanks for your consistent support about this package.

When I use this package in ROS2, I will try the way you mentioned.

Thank you :)

closed 2020-11-12 by SeungRyeol