carlosmccosta/dynamic_robot_localization #22: How to use 6 DoF robot localization

SeungRyeol, 2020-11-12:
I followed your instructions and applied "3 DoF robot localization / mapping" to my real robot successfully. (https://github.com/carlosmccosta/dynamic_robot_localization/issues/18#issuecomment-640266635)

And I want to try 6 DoF robot localization now.

First, I wonder if I must use this file(https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/labiomep/labiomep_6dof.launch).

Or should I use octo_map.launch?

Can you tell me which parameters to change and which launch file to use?

carlosmccosta, 2020-11-13:
Good night,

For 6 DoF tracking you can use as a starting point the launch files from [labiomep_6dof.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/labiomep/labiomep_6dof.launch) or [ethzasl_kinect_dataset.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/asl/ethzasl_kinect_dataset.launch) (mainly to see the yaml configurations for the drl pipeline, the rest is for configuring the test environment and automatically load the bag files).

[OctoMap](http://octomap.github.io) can be used for adding the mapping functionality with ray tracing and dynamic addition and removal of geometry. 

drl can do additive mapping internally (it can keep adding the registered point clouds to its reference map, if you enable that feature). But if you require dynamic maps in which there are dynamic objects moving around the 3D space, octomap allows the removal of these points, because it does ray tracing -> when several sensor rays from consecutive scans pass through voxels that were marked as occupied (using data from previous sensor scans), then, those 3D cells are marked as "free space", effectively removing geometry / points from the map (because they no longer occupy that 3D space).

But in the weekend I will make a more user friendly launch file within the [dynamic_robot_localization launch folder](https://github.com/carlosmccosta/dynamic_robot_localization/tree/melodic-devel/launch).
The [drl pipeline is highly reconfigurable](https://github.com/carlosmccosta/dynamic_robot_localization/blob/melodic-devel/yaml/schema/drl_configs.yaml), so you can optimize it further to your particular use case. Nevertheless, I will add a generic 6 DoF tracking launch, so you can have a more user friendly starting point.

Have a nice day :)

SeungRyeol, 2020-11-16:
Thanks for your detailed explanation. :)

I'm looking forward to your more launch file.

I will use your outdated launch file until your next update.

tenneck, 2020-11-23:
Hi Carlos,
I also wanted to mention, that I'm looking forward to a more generic 6 dof tracking launch file.
Keep up the great work!
- Tobias

carlosmccosta, 2020-11-24:
Hello,

I am occupied with other research projects right now, but I might be able to allocate some time during the end of this week to create the launch file after I finish the implementation of the Meijster distance transform in my PCL fork for initializing the [correspondence estimation using look up tables](https://github.com/carlosmccosta/pcl/blob/master-all-pr/registration/include/pcl/registration/correspondence_estimation_lookup_table.h) that I implemented in my PCL fork -> [it can significantly speedup ICP](https://www.researchgate.net/publication/322664340_Map-Matching_Algorithms_for_Robot_Self-Localization_A_Comparison_Between_Perfect_Match_Iterative_Closest_Point_and_Normal_Distributions_Transform).

Have a nice day,

tenneck, 2020-11-26:
Great, I'm looking forward to it. The results in your paper look very promising, too. That could result in much better real-time 6dof localization.

WilliamWoo45, 2020-12-24:
Hi Carlos! @carlosmccosta

In recent days, I'm trying to implement your **3-D (6 DoF) Global Localization** approach to achieve `initial pose estimation` and `pose recovery`. I'm wondering:
1. Which `launch `file should I use?
2. Is there any recommended `demo dataset` that I can use for testing?
3. Should I follow similar steps as the previous [2D localization case you guided me](https://github.com/carlosmccosta/dynamic_robot_localization/issues/18#issuecomment-644139778) ?

Thanks again!
Wish you & your family a Merry Christmas and Happy New Year ! :)

Best Regards,
William

carlosmccosta, 2020-12-24:
Hello,

As a starting point you can look at [labiomep_6dof.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/labiomep/labiomep_6dof.launch) and [ethzasl_kinect_dataset.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/asl/ethzasl_kinect_dataset.launch), which have both the configurations for drl and datasets.

I haven't had time yet to finish the lookup tables initialization with the Meijster algorithm. When I finish implementing it, I will create a more generic 6 DoF launch file using lookup tables for faster localization (when compared with kd-trees).

Merry Christmas and happy new year :)

WilliamWoo45, 2020-12-28:
Dear Carlos, 

Thanks for your speedy reply. 
I'm very looking forward to your more generic 6 DoF launch file :)

Best Regards,
William

tenneck, 2021-02-09:
Hi Carlos,

I wonder whether you had success implementing and testing the lookup tables initialization with the Meijster algorithm. Any news on that topic?

Best, T

carlosmccosta, 2021-02-09:
Good night,

I am involved in other research projects and haven't had time to finish the Meijster lut initialization.
Nevertheless, the lut initialization with kd-trees is implemented, but it takes longer to initialize the lut when compared with the Meijster algorithm.
I may have some time next month, depending on how long it takes to finish the other projects tasks.

Have a nice day :)

carlosmccosta, 2021-04-03:
Hello,

I refactored [ethzasl_kinect_dataset.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/hydro-devel/launch/environments/asl/ethzasl_kinect_dataset.launch) into a more generic 6 dof launch file available at [dynamic_robot_localization_system_6dof.launch](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch).

Examples on how to use it in [use_cases.md](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/docs/use_cases.md#6-dof-tracking).

For now, the usage of lookup tables (LUTs) in 6 DoF requires the initialization with [k-d trees](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_3d.yaml#L61) (when I have some time I will finish the Meijster algorithm to speedup its initialization).

For activating the LUTs it is only necessary to change the `correspondence_estimation_approach` from `CorrespondenceEstimation` to `CorrespondenceEstimationLookupTable` in the [cluttered_environments_dynamic_large_map_3d.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_3d.yaml#L55).

The lookup tables can speedup the correspondence estimation phase of the iterative closest point (ICP) algorithm from 20 to 60 times, depending on the sensor data, reference point cloud and configurations of drl.

But keep in mind that a LUT with small [cell size](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_3d.yaml#L57) (below 1 cm) for large reference point clouds (bounding box with several meters) requires a lot of RAM, while the corresponding k-d tree is extremely efficient in memory usage (at the cost of being slower than the LUT).

Also, k-d trees by being sparse search data structures initialize much faster than LUTs.

Have a nice day :)

closed 2021-04-13 by SeungRyeol