carlosmccosta/dynamic_robot_localization #27: Mismatch between 6DoF tracking achieved locally and the reference video with same data set

d1504, 2021-03-12:
Hi,
I am trying to reproduce 6DoF tracking as in Video 6: Free fly test with Kinect in the ETHZ RGB-D dataset using the 6 DoF localization system.
After configuring workspace and installing all packages, I have executed following commands from terminal.
$ source ~/catkin_ws_drl/devel_release/setup.bash 
$ cd /catkin_ws_drl/src/dynamic_robot_localization_tests/launch/environments/asl/bags
$ roslaunch ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch

Once RViz comes up, I switched to terminal and pressed space bar.
Now, tracking gets visualized in RViz.
However it does not look like that in reference video (https://www.youtube.com/watch?v=UslDiUkm7wE)

The output I got is shared here: https://youtu.be/8FZZhAnhFvs
**What could be the reason for this mismatch?** 

I am also getting **set of warnings and errors in terminal** as below.
[ WARN] [1615551179.209201725, 1298112281.916861620]: No transform between frames map_ground_truth_corrected and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.
[ WARN] [1615551179.213807044, 1298112281.916861620]: No transform between frames map and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.
[ WARN] [1615551179.222639773, 1298112281.916861620]: No transform between frames map_drl and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.
[ WARN] [1615551179.231428189, 1298112281.916861620]: No transform between frames odom and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.

[ WARN] [1615551190.198542742, 1298112287.954772391]: Discarded cloud because localization couldn't be calculated
[ WARN] [1615551190.198679600, 1298112287.954772391]: Failed point cloud processing with error [PointCloudFilteringFailed]
[ WARN] [1615551191.853380178, 1298112289.606053195]: Failed point cloud processing with error [PointCloudFilteringFailed]
[ WARN] [1615551191.886966564, 1298112289.636235797]: Discarded cloud because localization couldn't be calculated
[ WARN] [1615551191.886997556, 1298112289.636235797]: Failed point cloud processing with error [PointCloudFilteringFailed]
[ERROR] [1615551191.919287026, 1298112289.666394334]: Lost tracking!

In initial attempts, I got **warnings related to euclidean_transformation_validator parameters**, like:
[ WARN] [1615551455.183423693, 1298112283.032384128]: EuclideanTransformationValidator rejected new pose at time 1298112283.032384128 due to max_outliers_percentage -> 
	 correction translation: 3.12924 | (max_transformation_distance: 4)
	 correction rotation: 2.03959 | (max_transformation_angle: 0.1)
	 new pose diff translation: 3.12924 | (max_new_pose_diff_distance: 3.9)
	 new pose diff rotation: 2.03959 | (max_new_pose_diff_angle: 3.57)
	 root_mean_square_error: 0.0333355 | (max_root_mean_square_error: 0.05)
	 root_mean_square_error_reference_pointcloud: 0 | (max_root_mean_square_error_reference_pointcloud: -1)
	 outliers_percentage: 0.925824 | (max_outliers_percentage: 0.6)
	 outliers_percentage_reference_pointcloud: 0 | (max_outliers_percentage_reference_pointcloud: -1)
	 inliers_angular_distribution: 2 | (min_inliers_angular_distribution: 0.125)
	 outliers_angular_distribution: -2 | (max_outliers_angular_distribution: 0.875)
[pcl::IterativeClosestPointWithNormals::computeTransformation] Not enough correspondences found. Relax your threshold parameters.

Fo fix this, I had modified values in ../dynamic_robot_localization/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_3d.yaml as below:
transformation_validators:
    euclidean_transformation_validator:
        max_transformation_angle: 3.20
        max_transformation_distance: 4.0
        max_new_pose_diff_angle: 3.57
        max_new_pose_diff_distance: 3.90
        max_root_mean_square_error: 0.05
        max_outliers_percentage: 1.00

Please guide me to resolve this.

My configuration is:
- dynamic_robot_localization: melodic-devel
- dynamic_robot_localization_tests: hydro-devel
- Ubuntu 18.04
- Intel® Core™ i7-10810U CPU @ 1.10GHz × 12
- Quadro P520/PCIe/SSE2 Graphics
- 16 GB RAM
- 512 GB HDD

carlosmccosta, 2021-03-15:
Hello,

Thank you for reporting the problem :)

By mistake, I committed the usage of lookup tables for 6 DoF tracking with the Meijster algorithm for initialization of the lut.
I haven't had time to finish the implementation of the Meijster algorithm in my pcl fork, so if you want to use it, you can use kdtrees instead of the distance transform algorithm.
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/schema/drl_configs.yaml#L531
With lookup tables the icp registration algorithm will be faster when performing point cloud registration (when compared with the default kdtree search approach, for finding correspondences).

For having similar results as the video I uploaded to youtube, I made this change:
https://github.com/carlosmccosta/dynamic_robot_localization/commit/aeae8befd61dd33dbd30405fc729221e767a20e0

For testing the 6 DoF tracking, please revert the modifications you made.
`roslaunch dynamic_robot_localization_tests ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch`

The `FullIntegration` is an operation mode for slam, in which the full sensor point cloud is integrated / accumulated.
drl can perform 6 DoF tracking in real time.
But for performing slam in real time you need a very powerful cpu or you need to play the rosbag at a much lower rate, which was what I did to create the reference point cloud using drl.
https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/noetic-devel/launch/environments/asl/ethzasl_kinect_dataset.launch#L10
https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/noetic-devel/maps/asl/ethzasl_kinect_dataset/high-complexity-environment_icp-point-point_0.0025-voxel-grid_mls_0.005-rosbag-speed.ply

During tests for other use cases ([3D perception](https://github.com/carlosmccosta/object_recognition)), I changed the loading of the reference point cloud to happen before the setup of the ros publishers.

That was why you could not see the reference point cloud in rviz.
With the commit below you can now see the reference point cloud in rviz after its loaded from the .ply file by drl.
https://github.com/carlosmccosta/dynamic_robot_localization/commit/a79c09f5c81404e7eb6bf11b81cd4e79635d805c

For testing, please update all repositories to the latest commit of the most recent branch.
Namely, noetic-devel for:
https://github.com/carlosmccosta/dynamic_robot_localization
https://github.com/carlosmccosta/dynamic_robot_localization_tests
https://github.com/carlosmccosta/robot_localization_tools

And melodic-devel for:
https://github.com/carlosmccosta/pose_to_tf_publisher

Also, make sure that drl has loaded the reference point cloud before pressing the space key to play the rosbag (by looking at topic `/dynamic_robot_localization/reference_pointcloud` in rviz or command line).

Have a nice day,

d1504, 2021-03-16:
Hi Carlos, 

Thank you very much for a quick fix and detailed feedback. The reported issue is solved now. 

~ Divya

d1504, 2021-03-16:
Hi Carlos,

Thank you very much for the quick fix and detailed feedback. The reported issue is solved now.

~ Divya

closed 2021-03-16 by d1504