carlosmccosta/dynamic_robot_localization #30: compile errors

SU1997, 2021-07-06:
Hello Carlos Mccosta :)
when i compile from the source, some error occured:

Errors     << dynamic_robot_localization:make /home/curry/catkin_ws_drl/logs_release/dynamic_robot_localization/build.make.000.log                                                           
In file included from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/cloud_viewer.cpp:10:0:
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/cloud_viewer.hpp: In instantiation of ‘void dynamic_robot_localization::CloudViewer<PointT>::setupConfigurationFromParameterServer(ros::NodeHandlePtr&, ros::NodeHandlePtr&, const string&) [with PointT = pcl::PointXYZRGBNormal; ros::NodeHandlePtr = boost::shared_ptr<ros::NodeHandle>; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/cloud_viewer.cpp:19:1:   required from here
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/cloud_viewer.hpp:33:85: warning: unused parameter ‘node_handle’ [-Wunused-parameter]
 void CloudViewer<PointT>::setupConfigurationFromParameterServer(ros::NodeHandlePtr& node_handle, ros::NodeHandlePtr& private_node_handle, const std::string& configuration_namespace) {
                                                                                     ^~~~~~~~~~~
In file included from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/pointcloud_conversions.h:32:0,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/pointcloud_conversions.hpp:9,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/pointcloud_conversions.cpp:10:
/usr/include/pcl-1.8/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^~~~~~~~
In file included from /usr/include/c++/7/memory:80:0,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/pointcloud_conversions.h:12,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/pointcloud_conversions.hpp:9,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/pointcloud_conversions.cpp:10:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43:0:
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp: In instantiation of ‘bool dynamic_robot_localization::RegistrationVisualizer<PointSource, PointTarget>::setRegistration(pcl::Registration<PointSource, PointTarget>&) [with PointSource = pcl::PointXYZRGBNormal; PointTarget = pcl::PointXYZRGBNormal]’:
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:52:1:   required from here
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:70:2: error: no matching function for call to ‘pcl::Registration<pcl::PointXYZRGBNormal, pcl::PointXYZRGBNormal, float>::registerVisualizationCallback(std::function<void(const pcl::PointCloud<pcl::PointXYZRGBNormal>&, const std::vector<int>&, const pcl::PointCloud<pcl::PointXYZRGBNormal>&, const std::vector<int>&)>&)’
  registration.registerVisualizationCallback(this->update_visualizer_);
  ^~~~~~~~~~~~
In file included from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/registration_visualizer.h:41:0,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:39,
                 from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43:
/usr/include/pcl-1.8/pcl/registration/registration.h:375:7: note: candidate: template<class FunctionSignature> bool pcl::Registration<PointSource, PointTarget, Scalar>::registerVisualizationCallback(boost::function<FunctionSignature>&) [with FunctionSignature = FunctionSignature; PointSource = pcl::PointXYZRGBNormal; PointTarget = pcl::PointXYZRGBNormal; Scalar = float]
       registerVisualizationCallback (boost::function<FunctionSignature> &visualizerCallback)
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/registration/registration.h:375:7: note:   template argument deduction/substitution failed:
In file included from /home/curry/catkin_ws_drl/src/dynamic_robot_localization/src/common/registration_visualizer.cpp:43:0:
/home/curry/catkin_ws_drl/src/dynamic_robot_localization/include/dynamic_robot_localization/common/impl/registration_visualizer.hpp:70:2: note:   ‘std::function<void(const pcl::PointCloud<pcl::PointXYZRGBNormal>&, const std::vector<int>&, const pcl::PointCloud<pcl::PointXYZRGBNormal>&, const std::vector<int>&)>’ is not derived from ‘boost::function<Signature>’
  registration.registerVisualizationCallback(this->update_visualizer_);

i dont kown the reasons,could you please help me to solve it? thanks again

carlosmccosta, 2021-07-06:
Hello,

In the log it shows pcl 1.8.
For compiling drl you need to compile [my fork of pcl](https://github.com/carlosmccosta/pcl) first, which is based on pcl 1.11.1
After compiling and sourcing the catkin `setup.bash`, when you do `roscd pcl`, it should go to the source folder of my pcl fork.

Check the install instructions here:
[https://github.com/carlosmccosta/dynamic_robot_localization#installation-from-source](https://github.com/carlosmccosta/dynamic_robot_localization#installation-from-source)

Have a nice day :)

SU1997, 2021-07-08:
> Hello,
>
> In the log it shows pcl 1.8.
> For compiling drl you need to compile [my fork of pcl](https://github.com/carlosmccosta/pcl) first, which is based on pcl 1.11.1
> After compiling and sourcing the [...]

hi sorry to  trouble you again,i finished the compile problem.and when i used the commend:
`roslaunch ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch`
something error happend:
[ERROR] [1625708402.569137124, 1298112281.916861620]: Reference point cloud topic or file for localization system must be provided!
[ WARN] [1625708402.634751421, 1298112281.916861620]: No transform between frames map_ground_truth_corrected and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.
[ WARN] [1625708402.648442547, 1298112281.916861620]: No transform between frames map and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.
[ WARN] [1625708402.659599306, 1298112281.916861620]: No transform between frames map_drl and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.
[ WARN] [1625708402.664855349, 1298112281.916861620]: No transform between frames odom and base_link available after 1298112281.916862 seconds of waiting. This warning only prints once.

i have downloaded the ethzasl_kinect_dataset and put it in correct DIR
thanks:)

carlosmccosta, 2021-07-08:
Hello,

When I run:
```
roslaunch dynamic_robot_localization_tests ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch
```

And after a few seconds press space for triggering the rosbag play, the drl system performs like this video:
https://www.youtube.com/watch?v=UslDiUkm7wE

The ethzasl bags can be downloaded from:
http://www.files.ethz.ch/asl/ethzasl_kinect_dataset.tar.bz2

And the .bag files (such as 0high-0slow-0fly-0_2011-02-19-11-44-41.bag) should be extracted and placed inside this folder:
```
dynamic_robot_localization_tests/datasets/asl/ethzasl_kinect_dataset/
```

The map file `high-complexity-environment_icp-point-point_0.0025-voxel-grid_mls_0.005-rosbag-speed.ply` should be inside folder `dynamic_robot_localization_tests/maps/asl/ethzasl_kinect_dataset/`

Have a nice day,

SU1997, 2021-07-09:
> Hello,
>
> When I run:
>
> ```
> roslaunch dynamic_robot_localization_tests ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch
> ```
>
> And after a few seconds press space for triggering the rosbag [...]

hi carlos
i think i found the reason
i found that after used the function:
`bool Localization<PointT>::s_applyCloudFilters(std::vector< typename CloudFilter<PointT>::Ptr >& cloud_filters, typename pcl::PointCloud<PointT>::Ptr& pointcloud, int minimum_number_of_points_in_ambient_pointcloud)`
the size of pointcloud turned to zero, it may be the reason of my problem,what can i do to fix it?

carlosmccosta, 2021-07-10:
Hello,

That is strange.
In my laptop it is working as expected (the `s_applyCloudFilters` does not reduce the point cloud to zero points).

Are you running Ubuntu 20.04 and ROS Noetic with all the latest updates installed ?

Do you have the git repositories with these branches checked out ?

```
wstool info
workspace: /home/carloscosta/catkin_ws_drl/src

 Localname                  S SCM Version (Spec)     UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                  - --- --------------     -----------  ---------------------------
 pose_to_tf_publisher         git melodic-devel  (-) 5484c247216b github.com/carlosmccosta/pose_to_tf_publisher.git
 perception_pcl               git melodic-devel  (-) bd21be2cb8ec github.com/ros-perception/perception_pcl.git
 pcl_msgs                     git noetic-devel  (-)  ddcc8e14729f github.com/ros-perception/pcl_msgs.git
 pcl                          git master-all-pr  (-) e28ae1b6b813 github.com/carlosmccosta/pcl.git
 mesh_to_pointcloud           git master  (-)        8fbdaedfbe99 github.com/carlosmccosta/mesh_to_pointcloud.git
 laserscan_to_pointcloud      git master  (-)        31ab4f966202 github.com/carlosmccosta/laserscan_to_pointcloud.git
 dynamic_robot_localization   git noetic-devel  (-)  f786f18f7820 github.com/carlosmccosta/dynamic_robot_localization.git
```

You can also change the `ros_verbosity_level` to DEBUG for trying to identify which filter is causing the issue.
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L25

Keep in mind that you need to wait around 10 seconds after starting `roslaunch dynamic_robot_localization_tests ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch` for letting the system perform the setup. Then you can press the space key in the terminal to trigger the rosbag play.

Have a nice day,

SU1997, 2021-07-12:
> Hello,
>
> That is strange.
> In my laptop it is working as expected (the `s_applyCloudFilters` does not reduce the point cloud to zero points).
>
> Are you running Ubuntu 20.04 and ROS Noetic with all the [...]

hi thank you for your help
there is a new problem:
`[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.`
where can i change the parameters in the source code to fix it?

SU1997, 2021-07-12:
[image]
and this is the rviz shown

carlosmccosta, 2021-07-12:
Hello,

From the gif you seem to have modified the filters.
They are important to control the level of detail of the point cloud for allowing realtime 6 DoF pose.
For using the full kinect point cloud you would need a very powerful cpu.

For reference, I have an Intel i7-6700HQ, and drl runs well with the configurations that are in the [dynamic_robot_localization_tests](https://github.com/carlosmccosta/dynamic_robot_localization_tests) git.

Below is what the parameter server is supposed to have for configuring the drl pipeline, for allowing you to achieve the results shown in this video:
https://www.youtube.com/watch?v=UslDiUkm7wE

Which was also confirmed by other users with different hardware:
https://github.com/carlosmccosta/dynamic_robot_localization/issues/2#issuecomment-800182277

Have a nice day,

```
rosparam get / -p
dynamic_robot_localization:
  drl_localization_node:
    filters:
      ambient_pointcloud:
        1_voxel_grid:
          downsample_all_data: true
          filter_limit_field_name: z
          filter_limit_max: 5.0
          filter_limit_min: -5.0
          filtered_cloud_publish_topic: 
          leaf_size_x: 0.02
          leaf_size_y: 0.02
          leaf_size_z: 0.02
          save_leaf_layout: false
        4_crop_box:
          box_max_x: 1.15
          box_max_y: 1.15
          box_max_z: 2.15
          box_min_x: -1.15
          box_min_y: -1.15
          box_min_z: 0.8
          box_rotation_pitch: 0.0
          box_rotation_roll: 0.0
          box_rotation_yaw: 0.0
          box_translation_x: 0.0
          box_translation_y: 0.0
          box_translation_z: 0.0
          filtered_cloud_publish_topic: 
          invert_selection: false
        5_random_sample:
          filtered_cloud_publish_topic: 
          invert_sampling: false
          number_of_random_samples: 425
      ambient_pointcloud_map_frame:
        4_crop_box:
          box_max_x: 3.5
          box_max_y: 1.0
          box_max_z: 1.5
          box_min_x: 1.5
          box_min_y: -2.0
          box_min_z: -0.2
          box_rotation_pitch: 0.0
          box_rotation_roll: 0.0
          box_rotation_yaw: 0.0
          box_translation_x: 0.0
          box_translation_y: 0.0
          box_translation_z: 0.0
          filtered_cloud_publish_topic: ambient_pointcloud_filtered
          invert_selection: false
      reference_pointcloud:
        1_voxel_grid:
          downsample_all_data: true
[... 218 lines cut ...]
    odom_frame_id: odom
    odometry_topic: 
    pose_stamped_topic: 
    pose_with_covariance_stamped_topic: 
    poses_filename: 
    publish_initial_pose: true
    publish_last_pose_tf_timeout_seconds: -1.0
    publish_rate: -30.0
    tf_lookup_timeout: 0.5
    tf_time_offset: 0.0
    transform_pose_to_map_frame_id: true
    transform_tf_message_source: 
    transform_tf_message_target: 
rosdistro: |
  noetic
  
roslaunch:
  uris:
    host_13cmcc37__34511: http://13CMCC37:34511/
rosversion: |
  1.15.11
  
run_id: ebc80af4-e312-11eb-8b02-f59c60209de6
use_sim_time: true
```

carlosmccosta, 2021-07-12:
For more documentation about the drl pipeline, check the file drl_configs.yaml
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/schema/drl_configs.yaml

All configurations are loaded from yaml files into the parameter server using [launch files](https://github.com/carlosmccosta/dynamic_robot_localization/tree/noetic-devel/launch).
You do not need to edit any cpp source code (there aren't any hardcoded values, only default values and they are all loaded from the parameter server at runtime).