carlosmccosta/dynamic_robot_localization #32: localization result unstable

ZOUJIASHUAI, 2022-02-22:
Hello @carlosmccosta  ,thank you for the code,I checked out some of the issues and get the code worked with my own machine(sick nano scan3),I did a little bit change of param,and publish the odom to the laser frame ID, so the tf tree is map -> odom -> laser,I use srf to calculate  lidar odom,which works fine with sick lidar.also I  test localization of mrpt iris_lama slam_toolbox and drl on my machine,and I found that drl can recover when it lost,which seems more advanced ,but also have some unexpected result.

why I`m doing this?compare all this localization algorithm in my machine on same circumstances,because I have seem the localization algorithm of Bosch Rexroth,which can work on high dynamic environment with accurate result,and I can not get my machine work  on this environment.

expect : get stable pose result(±1cm) 
fact: pose result unstable(±4cm)
operation: pubish odom\publish scan\roslaunch dynamic_robot_localization_system.launch\set min_number_of_scans_to_assemble_per_cloud 1\set max_number_of_scans_to_assemble_per_cloud 1\file :filters_large_map_2d.yaml  number_of_random_samples: 1000\map_cell_resolution: 0.02

ZOUJIASHUAI, 2022-02-22:
bag file :https://drive.google.com/file/d/11hs3RYYo0G0OEeVuVGtgmWHdTxMGSUvw/view?usp=sharing
test video :https://drive.google.com/file/d/1ZNdq8uA8gfTsMU4QnMPhctZAzub2A1mN/view?usp=sharing
test video2:https://drive.google.com/file/d/1R_lI6Jw7V9tJlUYy2vIOuV2_bGwZwyov/view?usp=sharing

ZOUJIASHUAI, 2022-02-22:
mrpt :pose result(±8mm) can not recover when it lost ,no warning when it lost
iris lama:pose result(±8mm) can not recover when it lost,no warning when it lost
slam_toolbox: pose result drift some time,no warning when it lost
drl:pose result(±40mm) can recover when it lost,no result when it lost

carlosmccosta, 2022-05-12:
Hello :)

I am sorry, I only noticed this issue now.

Unstable localization usually happens due to:
- Odometry with low accuracy, drift or sudden jumps
  - Suggestion: recalibrate odometry or try without odometry, with a static tf between odom and base_footprint
- Map with filled walls which causes the matching algorithm to match inside the walls
  - Suggestion: edit the map (in gimp or using opencv) and ensure that the walls are composed of a single pixel layer, one for each side of the wall | alternatively, increase the voxel grid size used in drl for the map and eventually, the sensor data in the filters yaml (after changing the yaml, reload drl and check the parameter server to ensure that the values changed and check the drl reference point cloud in rviz). This will likely reduce the localization accuracy, but it will increase its stability
    - https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L26
    - https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/filters/filters_large_map_2d.yaml#L5
    - https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/filters/filters_large_map_2d.yaml#L16
    - https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/schema/drl_configs.yaml#L86

For filtering the drl pose, the use_ukf_filtering can be enabled, but the result is highly dependent on the covariance estimation and configuration of the kalman filter.
- https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L20
- https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/covariance_estimation/covariance_estimation.yaml
- https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/ukf.launch
- https://github.com/cra-ros-pkg/robot_localization

Have a nice day,