carlosmccosta/dynamic_robot_localization #33: 2D localization mode have roll pitch and Zaxis drit

ZOUJIASHUAI, 2022-08-01:
[image]

ZOUJIASHUAI, 2022-08-01:
@carlosmccosta the localization topic also seems have Zaxis drift.

carlosmccosta, 2022-08-01:
Hello,

For 3 DoF localization (x, y, yaw), you must use the [dynamic_robot_localization_system.launch](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch), which is configured for using 2D lidars and 2D occupancy grids.

drl assumes that your lidar is mounted parallel to the floor and after transforming the lidar data to the odom TF, it will set the points Z coordinate to zero.

If your lidar is not parallel to the floor, you will not be able to use 3 DoF localization algorithms and maps, because they all assume sensors parallel to the floor to ensure that when the sensor changes place within the map, the measurements remain consistent.

Have a nice day,

ZOUJIASHUAI, 2022-08-08:
@carlosmccosta 
hello,
I do use the [dynamic_robot_localization_system.launch],and below is the launch parameter.
<?xml version="1.0" encoding="UTF-8"?>
<launch>
	
	
	<arg name="pose_topic" default="initial_pose" />
	<arg name="pose_stamped_topic" default="initial_pose_stamped" />
	<arg name="pose_with_covariance_stamped_topic" default="/initialpose" /> 
	<arg name="pose_with_covariance_stamped_publish_topic" default="localization_pose_with_covariance" /> 
	
	
	<arg name="pose_stamped_publish_topic" default="localization_pose" />
	
	
	<arg name="map_frame_id" default="map" />
	<arg name="odom_frame_id" default="odom" />
	<arg name="base_link_frame_id" default="laser" />
	<arg name="sensor_frame_id" default="laser" />
	
	
	<arg name="use_ukf_filtering" default="false" />
	<arg name="nodes_namespace" default="super_slam" />
	<arg name="nodes_respawn" default="true" />
	<arg name="localization" default="localization" />
	<arg name="pcl_verbosity_level" default="ALWAYS" /> 
	<arg name="ros_verbosity_level" default="FATAL" /> 
	<arg name="yaml_configuration_filters_filename" default="$(find super_slam)/yaml/configs/filters/filters_large_map_2d.yaml" />
	<arg name="yaml_configuration_pose_estimation_filename" default="$(find super_slam)/yaml/configs/empty.yaml" />
	<arg name="yaml_configuration_tracking_filename" default="$(find super_slam)/yaml/configs/pose_tracking/cluttered_environments_dynamic_fast_2d.yaml" />
	<arg name="yaml_configuration_recovery_filename" default="$(find super_slam)/yaml/configs/pose_recovery/recovery_slam_2d.yaml" />
	<arg name="yaml_configuration_covariance_filename" default="$(find super_slam)/yaml/configs/covariance_estimation/covariance_estimation.yaml" if="$(arg use_ukf_filtering)" />
	<arg name="yaml_configuration_covariance_filename" default="$(find super_slam)/yaml/configs/empty.yaml" unless="$(arg use_ukf_filtering)" />
	
	
	<arg name="reference_pointcloud_filename" default="" />
	<arg name="reference_pointcloud_preprocessed_save_filename" default="" />
	<arg name="reference_pointcloud_available" default="true" />
	<arg name="reference_pointcloud_type_3d" default="false" />
	<arg name="reference_pointcloud_type" default="2D" unless="$(arg reference_pointcloud_type_3d)"/>
	<arg name="reference_pointcloud_type" default="3D" if="$(arg reference_pointcloud_type_3d)" />
	<arg name="reference_pointcloud_update_mode" default="NoIntegration" if="$(arg reference_pointcloud_available)"/> 
	<arg name="reference_pointcloud_update_mode" default="FullIntegration" unless="$(arg reference_pointcloud_available)"/>
	<arg name="reference_pointcloud_keypoints_filename" default="" />
	<arg name="reference_pointcloud_keypoints_save_filename" default="" />
	<arg name="reference_pointcloud_descriptors_filename" default="" />
	<arg name="reference_pointcloud_descriptors_save_filename" default="" />
	<arg name="reference_costmap_topic" default="" if="$(arg reference_pointcloud_type_3d)" />
	<arg name="reference_costmap_topic" default="/map" unless="$(arg reference_pointcloud_type_3d)" />
	<arg name="reference_costmap_topic_updated" default="/map_updated" />
	<arg name="reference_pointcloud_topic" default="reference_pointcloud_updated" if="$(arg reference_pointcloud_type_3d)" /> 
	<arg name="reference_pointcloud_topic" default="" unless="$(arg reference_pointcloud_type_3d)" />
	
	
	<arg name="ambient_pointcloud_topic" default="ambient_pointcloud" />
	<arg name="ambient_pointcloud_topic_disabled_on_startup" default="false" />
	<arg name="imu_topic" default="/imu/data" />
	
	
	<arg name="robot_initial_pose_in_base_to_map" default="false" /> 
	<arg name="robot_initial_pose_available" default="false" />
	<arg name="robot_initial_x" default="0.0" />
	<arg name="robot_initial_y" default="0.0" />
	<arg name="robot_initial_z" default="0.0" />
	<arg name="robot_initial_roll" default="0.0" />
	<arg name="robot_initial_pitch" default="0.0" />
	<arg name="robot_initial_yaw" default="0.0" />
	
	
[... 172 lines cut ...]
			<arg name="cloud_in" default="$(arg octomap_pointcloud_topic)" />
			<arg name="map_frame_id" default="$(arg map_frame_id)" />
			<arg name="base_frame_id" default="$(arg base_link_frame_id)" />
			<arg name="sensor_frame_id" default="$(arg octomap_sensor_frame_id)" />
			<arg name="override_sensor_z" default="$(arg octomap_override_sensor_z)" />
			<arg name="override_sensor_z_value" default="$(arg octomap_override_sensor_z_value)" />
			<arg name="resolution" default="$(arg octomap_resolution)" />
			<arg name="minimum_amount_of_time_between_ros_msg_publishing" default="$(arg octomap_minimum_amount_of_time_between_ros_msg_publishing)" />
			<arg name="minimum_number_of_integrations_before_ros_msg_publishing" default="$(arg octomap_minimum_number_of_integrations_before_ros_msg_publishing)" />
			<arg name="cloud_out_topic" default="$(arg octomap_cloud_out_topic)" />
			<arg name="map_out_topic" default="$(arg reference_costmap_topic_updated)" if="$(arg use_octomap_only_to_build_occupancy_grid)" />
			<arg name="map_out_topic" default="$(arg reference_costmap_topic)" unless="$(arg use_octomap_only_to_build_occupancy_grid)" />
			<arg name="nodes_respawn" default="$(arg nodes_respawn)" />
		</include>

		<group if="$(arg reference_pointcloud_available)" >
			<node name="map_server" pkg="map_server" type="map_server" args="$(arg initial_2d_map_file)" respawn="$(arg nodes_respawn)" clear_params="true" output="screen" unless="$(arg reference_pointcloud_type_3d)" >
				<param name="frame_id" type="str" value="$(arg map_frame_id)" />
				<remap from="map" to="initial_2d_map" />
			</node>
		</group>
	</group>

   <node pkg="rviz" type="rviz" name="rviz" args="-d /home/jason/agv_ws/dy.rviz"/>
</launch>

carlosmccosta, 2022-08-12:
Hello,

Check if the roll and pitch angles between the TFs odom and laser are 0 degrees.
The [2d icp configured in drl](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/include/dynamic_robot_localization/cloud_matchers/point_matchers/iterative_closest_point_2d.h#L41) assumes that the laser is parallel to the ground and uses the pcl [transformation_estimation_2D](
https://github.com/PointCloudLibrary/pcl/blob/master/registration/include/pcl/registration/transformation_estimation_2D.h)

Check your ROS parameter server (`rosparam get / -p`) to confirm that the 2d icp was correctly loaded.

Looking at your rviz screenshot it seems that the laser is tilted and not parallel to the ground.

Have a nice day :)

ZOUJIASHUAI, 2022-08-26:
hello @carlosmccosta 
this is my param,please check if anything wrong,I need a consecutive pose for navigation so I subscribed the tf (map->laser),I also found that the tf have x y yaw drift and increase the drift with time,but the pose from topic doesn't have drift,and the tf drift will be clean when drl received /initpose topic.

dynamic_robot_localization:
  drl_localization_node:
    cloud_analyzers:
      angular_distribution_analyzer:
        number_of_angular_bins: 270
      compute_inliers_angular_distribution: true
      compute_outliers_angular_distribution: true
    filters:
      ambient_pointcloud:
        1_voxel_grid:
          downsample_all_data: false
          filter_limit_field_name: z
          filter_limit_max: 2.0
          filter_limit_min: -1.0
          filtered_cloud_publish_topic: 
          leaf_size_x: 0.02
          leaf_size_y: 0.02
          leaf_size_z: 0.02
          save_leaf_layout: false
        5_random_sample:
          filtered_cloud_publish_topic: ambient_pointcloud_filtered
          invert_sampling: false
          number_of_random_samples: 2500
      reference_pointcloud:
        1_voxel_grid:
          downsample_all_data: false
          filter_limit_field_name: z
          filter_limit_max: 1.8
          filter_limit_min: -0.001
          filtered_cloud_publish_topic: 
          leaf_size_x: 0.02
          leaf_size_y: 0.02
          leaf_size_z: 0.02
          save_leaf_layout: false
    frame_ids:
      base_link_frame_id: laser
      map_frame_id: map
      odom_frame_id: odom
      sensor_frame_id: laser
    initial_pose:
      orientation_rpy:
        pitch: 0.0
        roll: 0.0
        yaw: 0.0
      position:
        x: 0.0
        y: 0.0
        z: 0.0
      robot_initial_pose_available: false
      robot_initial_pose_in_base_to_map: false
    initial_pose_estimators_matchers:
      feature_matchers:
        reference_pointcloud_descriptors_filename: 
        reference_pointcloud_descriptors_save_filename: 
    keypoint_detectors:
      reference_pointcloud:
        reference_pointcloud_keypoints_filename: 
        reference_pointcloud_keypoints_save_filename: 
    localization_point_type: PointXYZINormal
    message_management:
      tf_timeout: 0.1
    outlier_detectors:
      euclidean_outlier_detector:
        aligned_pointcloud_inliers_publish_topic: /aligned_pointcloud_inliers
        aligned_pointcloud_outliers_publish_topic: /aligned_pointcloud_outliers
        max_inliers_distance: 0.06
    pcl_verbosity_level: ALWAYS
[... 106 lines cut ...]
    float_update_field: 
    float_update_field_orientation_in_degrees: false
    initial_pitch: 0.0
    initial_pose_in_base_to_map: false
    initial_roll: 0.0
    initial_x: 0.0
    initial_y: 0.0
    initial_yaw: 0.0
    initial_z: 0.0
    invert_tf_hierarchy: false
    invert_tf_transform: false
    map_frame_id: map
    odom_frame_id: odom
    odometry_topic: 
    pose_stamped_topic: localization_pose
    pose_with_covariance_stamped_topic: /initialpose
    poses_filename: 
    publish_initial_pose: true
    publish_last_pose_tf_timeout_seconds: -3330.0
    publish_rate: 3000.0
    tf_lookup_timeout: 1.5
    tf_time_offset: 0.0
    transform_pose_to_map_frame_id: true
    transform_tf_message_source: 
    transform_tf_message_target:

carlosmccosta, 2022-08-28:
Hello,

What is your TF chain configuration from `map_frame_id` to `base_link_frame_id` ?

From your previous launch files, I assumed that it was `map -> odom -> laser`

If your odom is not accurate, you can disable its usage within `drl` by:

- Setting the [odom_frame_id](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L15) to a value equal to the [base_link_frame_id](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L16)
- Since a TF frame can only have one parent in the TF tree, for bypassing the `odom_frame_id`, it is necessary to:
  - Set to `true` the flag [invert_tf_transform](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L97)
  - Set to `true` the flag [invert_tf_hierarchy](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L98)
  - This way, the TF and [pose_stamped_publish_topic](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L11) will be published directly from [base_link_frame_id](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L16) to [map_frame_id](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L14) (the pose is specified in the `map_frame_id` coordinate system)
  - You can check the `drl` pose and TF in rviz with the fixed frame configured to `map` and can also check in the console with:
    - `rostopic echo /dynamic_robot_localization/localization_pose`
    - `rosrun tf tf_echo map laser _rate:=30 _precision:=4`

I rechecked one of my validation testes from a rosbag ([freiburg2_pioneer_slam1.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/noetic-devel/launch/environments/tum/bags/freiburg2_pioneer_slam1.launch)) and the latest commit of `drl` in the `noetic-devel` branch is performing as expected and without drift, as shown in this video:
https://www.youtube.com/watch?v=jAJ5wiN-mJ8

Have a nice day,

ZOUJIASHUAI, 2022-09-02:
hello @carlosmccosta 
I found another problem from drl,my odom is accurate so I read the map->laser tf to navigation,and the pose will jump to far a way if the map changed alot and the initial_pose_estimation is set,so I disabled the initial_pose_estimation.but some time it wouldn‘t be able to find  it self in the map even the environment is not so bad,I also tune some param from pose_recovery file,so does the initial_pose_estimation will help this situation？or how to tune the initial_pose_estimation so the pose will not jump to far?

ZOUJIASHUAI, 2022-09-02:
@carlosmccosta 
below is the param I set
************************initial_pose_estimation_2d***********************************************************
normal_estimators:
    reference_pointcloud:
        display_normals: false
        display_occupancy_grid_pointcloud: false
        flip_normals_using_occupancy_grid_analysis: true
        occupancy_grid_analysis_radius_resolution_percentage: 4.0
        normal_estimator_sac:
            model_type: 'SACMODEL_LINE'
            method_type: 'SAC_RANSAC'
            inlier_distance_threshold: 0.06
            max_iterations: 50
            sac_termination_probability: 0.99
            optimize_coefficients: true
            min_model_radius: -1.0
            max_model_radius: -1.0
            random_samples_max_k: 0
            random_samples_max_radius: 0.06
            minimum_inliers_percentage: 0.5
    ambient_pointcloud:
        compute_normals_when_tracking_pose: false
        compute_normals_when_recovering_pose_tracking: false
        compute_normals_when_estimating_initial_pose: true
        display_normals: false
        normal_estimator_sac:
            model_type: 'SACMODEL_LINE'
            method_type: 'SAC_RANSAC'
            inlier_distance_threshold: 0.06
            max_iterations: 50
            sac_termination_probability: 0.99
            optimize_coefficients: true
            min_model_radius: -1.0
            max_model_radius: -1.0
            random_samples_max_k: 0
            random_samples_max_radius: 0.065
            minimum_inliers_percentage: 0.5

curvature_estimators:
    reference_pointcloud:
        principal_curvatures_estimation:
            colorize_pointcloud_with_curvatures: true
            search_k: 0
            search_radius: 0.06
            curvature_type: 'CURVATURE_TYPE_MEAN'
            update_normals_with_principal_component_directions: false
    ambient_pointcloud:
        principal_curvatures_estimation:
            colorize_pointcloud_with_curvatures: true
            search_k: 0
            search_radius: 0.06
            curvature_type: 'CURVATURE_TYPE_MEAN'
            update_normals_with_principal_component_directions: false

keypoint_detectors:
    reference_pointcloud:
        sift_3d:
            min_scale: 0.015
            number_octaves: 4
            number_scales_per_octave: 8
            min_contrast: 0.0035
            sift3d_keypoints_cloud_publish_topic: 'reference_keypoints'
#        intrinsic_shape_signature_3d:
#            salient_radius: 0.14
#            non_max_radius: 0.08
#            normal_radius: 0.08
#            border_radius: 0.0
#            threshold21: 0.975
#            threshold32: 0.975
[... 232 lines cut ...]
#            invert_selection: false
#            filtered_cloud_publish_topic: ''
#        7_covariance_sampling:
#            number_of_samples: 100
#            filtered_cloud_publish_topic: ''
#        8_radius_outlier_removal:
#            radius_search: 1.0
#            min_neighbors_in_radius: 2
#            invert_removal: false
#            filtered_cloud_publish_topic: ''
****************************************************************
********************covariance_estimation.yaml***************
registration_covariance_estimator:
    error_metric: 'PointToPlane3D'
    correspondance_estimator: 'CorrespondenceEstimation'
    correspondence_distance_threshold: 0.1
    sensor_std_dev_noise: 0.015
    use_reciprocal_correspondences: false
    filtered_reference_cloud_publish_topic: 'covariance_reference_cloud'
    filtered_ambient_cloud_publish_topic: 'covariance_ambient_cloud'
    random_sample:
        number_of_random_samples: 300
        invert_sampling: false
        filtered_cloud_publish_topic: ''
******************************************************************

carlosmccosta, 2022-09-07:
Hello,

If drl is rejecting its pose estimation in your typical use case / environment, then you may need to increase the tolerance to outliers, by specifying to drl that the map data might be `X%` different in relation to the sensor data.

This is controlled with [max_inliers_distance](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L49) (in meters) and [max_outliers_percentage](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L68) (range 0 to 1 | note that currently this is an [outlier_ratio](https://github.com/carlosmccosta/dynamic_robot_localization/blob/efec43006d3d407bf62ef3f5dc55c12ecdd61627/include/dynamic_robot_localization/localization/impl/localization.hpp#L2495) -> 1 equals to 100%)

It is possible to control how far the tracking and recovery algorithms corrections translate and rotate the robot pose with [max_transformation_distance](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L64) | [max_transformation_angle](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L63) and for recovery [recovery_2d.yaml#L27](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_recovery/recovery_2d.yaml#L27) | [recovery_2d.yaml#L26](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_recovery/recovery_2d.yaml#L26)

For the initial pose algorithms, currently it not applied this jump limit to allow the robot to start lost and then find itself in the map.

To ensure that ICP runs fast, confirm that you have the look up table enabled with the [correspondence_estimation_approach](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L4) set to `CorrespondenceEstimationLookupTable` and with its [map_cell_resolution](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L6) equal to your map, otherwise the icp might stop its convergence after reaching its time limit defined in [convergence_time_limit_seconds](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L20)

If your robot is moving very fast, you may need to increase [max_correspondence_distance](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L21)

Have a nice day :)

ZOUJIASHUAI, 2022-10-29:
hello @carlosmccosta 
my odom is accurate，in odom topic I set 0 to angular_velocity and oritation in X Y axis,and set ignore_height_corrections: true, and it doesnt have roll pitch and Zaxis drit,but it still have x y pose drift and yaw drift,you can see the video below,with the increasement of running time ,the drift also increase,and the drift can be clear by send a initialpose topic ,seem the initialpose will triggle another program that recalculate the pose in the map and clear some drift,what I did is send initialpose every 60s ,but I think it can be fix in code which I can not found it,by the way I disabled the pose_estimation module,so maybe the initpose triggled the pose recovery module and cleared the drift?

https://drive.google.com/file/d/18HqVuU20vCfN-7rdiI9YQJneGD024vOo/view?usp=share_link
this video been record after the robot running 1 hours

carlosmccosta, 2022-10-29:
Hello,

When you send an initial pose, the matching algorithm changes from [yaml_configuration_tracking_filename](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L28) to [yaml_configuration_recovery_filename](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L29)

Assuming you have not changed the yamls loaded, you can compare the configurations here:
- [cluttered_environments_dynamic_large_map_2d.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml)
- [recovery_2d.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_recovery/recovery_2d.yaml)

You should not resend initial poses to fix this issue. Instead, you should adjust the configuration of drl.

Namely, try to increase these 3 parameters:
- [max_correspondence_distance](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L21)
- [convergence_time_limit_seconds](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L20)
- [max_number_of_registration_iterations](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L24)

You can also try and change the tracking algorithm from icp 2d to the standard icp, by [commenting this line](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L16) and [uncommenting this line](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L15) and then changing [transformation_estimation_approach](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L2) to `TransformationEstimationSVD`

Lastly, if it still has drift, you can disable the look up tables and use the standard k-d trees for correspondence estimation (they are slower, but you can try then and see if the problem persists). For that, change the [correspondence_estimation_approach](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_2d.yaml#L4) to `CorrespondenceEstimation`

Have a nice day,

ZOUJIASHUAI, 2022-10-31:
> CorrespondenceEstimation

hello @carlosmccosta 
I have copy [cluttered_environments_dynamic_large_map_2d.yaml]
 and [recovery_2d.yaml] to the drl and change the [correspondence_estimation_approach] in cluttered_environments_dynamic_large_map_2d.yaml to CorrespondenceEstimation,as you can see the link below,the problem still exist,I think this problem is not about the parameter,the point clould topic /dynamic_robot_localization/ambient_pointcloud and laserscan topic still drift slowly to map seems the drift are connect with the running distance,but /dynamic_robot_localization/aligned_pointcloud always aligned with map.maybe some TF code didn`t wrote properly?

https://drive.google.com/file/d/1AA0dVNP_Xgvrz4QMpX9hlSJ_GixUyJhv/view?usp=share_link

ZOUJIASHUAI, 2022-10-31:
> CorrespondenceEstimation

@carlosmccosta 
I found out that the /dynamic_robot_localization/aligned_pointcloud frame id is map,however the /dynamic_robot_localization/ambient_pointcloud frame id is odom,but the /dynamic_robot_localization/localization_detailed frame is map,

ZOUJIASHUAI, 2022-10-31:
> > CorrespondenceEstimation
> 
> @carlosmccosta I found out that the /dynamic_robot_localization/aligned_pointcloud frame id is map,however the /dynamic_robot_localization/ambient_pointcloud frame id is odom,but the /dynamic_robot_localization/localization_detailed frame is map,

@carlosmccosta 
looks like the accumulation error of odom has been add to loc pose,I set <arg name="assemble_lasers_on_map_frame" default="true" />, the drift problem seems to be fixed, but it came out a new problem,when I send initpose topic,the pose seems very unstable,and is very hard to let laser aligned with map at startup. and the pose is not stable as before when the robot is static.
https://drive.google.com/file/d/1Bf7CYHju0hmNpTgOpUr5S3t4NAqcuFPK/view?usp=share_link

carlosmccosta, 2022-10-31:
Hello,

The `ambient_pointcloud` topic should be assembled in the `odom` frame and not in the `map` frame, because `laser` -> `odom` provides a smooth tf chain, while `laser` -> `map` has discrete small jumps that are necessary and introduced to correct the odom drift over time. These jumps distort the point cloud assebly when the spherical interpolation is enabled. As such, the point cloud assemby should not be done on the `map` frame, otherwise, these distortions will cause major oscilations in the alignment performed by `drl`.

The `aligned_pointcloud` is the result of the alignment performed by drl and is on `map` frame, and as such, is the one you should be analyzing in rviz. You should not evaluate the `ambient_pointcloud` because that is the input for drl and not its result. And of course, since the `ambient_pointcloud` is on `odom` frame, it will show on rviz with the odometry drift. That is working as is supposed to.

These frames names are this way on purpose and follow the standard ros tf for mobile robotics:
https://www.ros.org/reps/rep-0105.html
https://navigation.ros.org/setup_guides/transformation/setup_transforms.html

Looking at your videos, the aligned_pointcloud is correctly overlapping the map. So drl was working correctly all along !
Keep in mind that every localization system has oscillations (which is different than drift) in its pose estimation due to lidar noise and also due to sliding of the alignment within the walls. Namely, if your map has walls with more than one pixel thickness in the occupancy grid, the alignment algorithms may oscillate. This is normal and if you want to minimize this oscillation, you need to have maps with 1 pixel thickness walls.

For inspecting the results of `drl`, set your rviz world frame to `map` and inspect the `aligned_pointcloud` overlap with the occupancy grid / map.
For checking the drl pose, you have the topics `localization_pose` and `localization_detailed`
Alternatively, you can check the pose between `map` and `base_link` or `base_footprint`

In [drl_configs.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/schema/drl_configs.yaml#L85) you can check the subscribe / input topics and the publish / output topics of `drl`

Have a nice day,

ZOUJIASHUAI, 2022-11-01:
> Hello,
>
> The `ambient_pointcloud` topic should be assembled in the `odom` frame and not in the `map` frame, because `laser` -> `odom` provides a smooth tf chain, while `laser` -> `map` has discrete [...]

hello @carlosmccosta 
I tried all the param you suggest,please watch this video,this is the robot ran 1 hours and stoped,the pose of localization_pose localization_detailed and the map->base_link is x=10.29 y=-15.39,then I send a initalpose at x=10.29 y=-15.39,and the pose of X Y axis has changed over 13cm after initpose to x = 10.42 y=-15.25 which is the correct pose to the real world,the map is 0.02 res,and I sure this is not because of oscillations .
https://drive.google.com/file/d/18HqVuU20vCfN-7rdiI9YQJneGD024vOo/view