carlosmccosta/dynamic_robot_localization #34: Pointclouds get dropped

oskarmue, 2022-08-12:
Thank you for your great work with this package, it looks really promising!
I recorded a 3d Pointcloud map of our environment beforhand. And now I'd like to lokalize our robot in said map.
Therefore I wanted to use the ynamic_robot_localization_system_6dof.launch launchfile. But when starting it, after the Map is loaded and global pose estimate is published, all the new Pointclouds get dropped, because:

[ WARN] [1660297010.079226327]: Dropping pointcloud because tf between base_link and odom isn't available
Failed to find match for field 'rgb'.
Failed to find match for field 'normal_x'.
Failed to find match for field 'normal_y'.
Failed to find match for field 'normal_z'.
Failed to find match for field 'curvature'.

Is that a Problem because of the pointcloud type I use? I am using a robosense Lidar and can decide between XYZIRT and XYZI type. So there is no option for rgb... . Am I still able to use this package somehow?

carlosmccosta, 2022-08-12:
Hello,

If you do not have odometry, then you need to set the [odom_frame_id](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch#L6) to be the frame as your [base_link_frame_id](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch#L7).

Have a nice day :)

carlosmccosta, 2022-09-07:
Hello,

That error seems to be from [sac_model_registration.h#L248](https://github.com/PointCloudLibrary/pcl/blob/3a408755361b6f108fa3e003013970780dea540f/sample_consensus/include/pcl/sample_consensus/sac_model_registration.h#L248), in which the covariance matrix is computed with [centroid.hpp#L508](https://github.com/PointCloudLibrary/pcl/blob/28e6312efab9e08cdd783d0c00ee46238c3bba55/common/include/pcl/common/impl/centroid.hpp#L508).

That class is used in drl in [cloud_matcher.hpp#L65](https://github.com/carlosmccosta/dynamic_robot_localization/blob/3b662e9d756ed34e0e2690d1c1aa5e63dec63e4d/include/dynamic_robot_localization/cloud_matchers/impl/cloud_matcher.hpp#L65) if the `max_number_of_ransac_iterations > 0`, which is the default for [recovery_3d.yaml#L12](https://github.com/carlosmccosta/dynamic_robot_localization/blob/3b662e9d756ed34e0e2690d1c1aa5e63dec63e4d/yaml/configs/pose_recovery/recovery_3d.yaml#L12) (you can set it to 0 to disable the usage of ransac within ICP) and also in [sample_consensus_prerejective.hpp#L542](https://github.com/carlosmccosta/dynamic_robot_localization/blob/3b662e9d756ed34e0e2690d1c1aa5e63dec63e4d/include/dynamic_robot_localization/cloud_matchers/feature_matchers/impl/sample_consensus_prerejective.hpp#L542)

drl removes NaN points from the point clouds when it receives sensor data ([localization.hpp#L1953](https://github.com/carlosmccosta/dynamic_robot_localization/blob/efec43006d3d407bf62ef3f5dc55c12ecdd61627/include/dynamic_robot_localization/localization/impl/localization.hpp#L1953)) and after normal estimation ([localization.hpp#L2875](https://github.com/carlosmccosta/dynamic_robot_localization/blob/efec43006d3d407bf62ef3f5dc55c12ecdd61627/include/dynamic_robot_localization/localization/impl/localization.hpp#L2875)).

As such, it is odd that the pcl function [computeMeanAndCovarianceMatrix](https://github.com/PointCloudLibrary/pcl/blob/28e6312efab9e08cdd783d0c00ee46238c3bba55/common/include/pcl/common/impl/centroid.hpp#L508) is generating a covariance matrix with NaN values.

Keep in mind that the [dynamic_robot_localization_system_6dof.launch](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch) does not have the initial pose estimation module loaded by default and as such, it expects the end user to specify the initial pose either in that launch file [dynamic_robot_localization_system_6dof.launch#L23](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch#L23) or given in a topic [dynamic_robot_localization_system.launch#L4](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L4)

Alternatively, you can enable the initial pose module by commenting the line with the empty.yaml ([dynamic_robot_localization_system_6dof.launch#L44](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch#L44)) and uncommenting the line below ([dynamic_robot_localization_system_6dof.launch#L45](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch#L45))

The 3d reference point cloud can be provided in [reference_pointcloud_filename](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system_6dof.launch#L11) or in [reference_pointcloud_topic](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L49).

I tested the 6 DoF tracking with the [ETHZ ASL dataset](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/noetic-devel/datasets/asl/ethzasl_kinect_dataset/rosbags.txt) using their high complexity environments rosbags. One of the tests can be started with [ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch](https://github.com/carlosmccosta/dynamic_robot_localization_tests/blob/noetic-devel/launch/environments/asl/bags/ethzasl_kinect_dataset_high_complexity_slow_fly_movement.launch) which should have results similar to [this video](https://www.youtube.com/watch?v=UslDiUkm7wE)

If your sensor data has a scale that is very different (objects much smaller or larger than the ones present in the video above), you may need to adjust the search radius / k / correspondence distance values of the files: [initial_pose_estimation_3d.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_estimation/initial_pose_estimation_3d.yaml), [cluttered_environments_dynamic_large_map_3d.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_tracking/cluttered_environments_dynamic_large_map_3d.yaml) and [recovery_3d.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/configs/pose_recovery/recovery_3d.yaml)

Have a nice day :)

closed 2022-10-24 by oskarmue