carlosmccosta/dynamic_robot_localization #35: error：registration.registerVisualizationCallback(this->update_visualizer_);

qqngwzlb, 2023-03-08:
error: no matching function for call to ‘pcl::Registration<pcl::PointXYZRGBNormal, pcl::PointXYZRGBNormal, float>::registerVisualizationCallback(std::function<void(const pcl::PointCloud<pcl::PointXYZRGBNormal>&, const std::vector<int>&, const pcl::PointCloud<pcl::PointXYZRGBNormal>&, const std::vector<int>&)>&)’
  registration.registerVisualizationCallback(this->update_visualizer_);

carlosmccosta, 2023-03-08:
Hello,

You need to use my fork of pcl when compiling drl:
- https://github.com/carlosmccosta/pcl

Check documentation for more information:
- https://github.com/carlosmccosta/dynamic_robot_localization#installation

Have a nice day,

43998213, 2023-03-09:
> Hello,
> 
> You need to use my fork of pcl when compiling drl:
> 
> * https://github.com/carlosmccosta/pcl
> 
> Check documentation for more information:
> 
> * https://github.com/carlosmccosta/dynamic_robot_localization#installation
> 
> Have a nice day,

env
pcl-pcl-1.8.1-cc-master-all-pr
dynamic_robot_localization-1.2.0

catkin build dynamic_robot_localization -j1 --verbose

[ 97%] Built target drl_localization
/usr/bin/make -f CMakeFiles/drl_localization_node.dir/build.make CMakeFiles/drl_localization_node.dir/depend
make[2]: 进入目录“/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization”
cd /home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/zjh/cocobot-six-dynamic-2/src/dynamic_robot_localization-1.2.0 /home/zjh/cocobot-six-dynamic-2/src/dynamic_robot_localization-1.2.0 /home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization /home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization /home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization/CMakeFiles/drl_localization_node.dir/DependInfo.cmake --color=
Dependee "/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization/CMakeFiles/drl_localization_node.dir/DependInfo.cmake" is newer than depender "/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization/CMakeFiles/drl_localization_node.dir/depend.internal".
Dependee "/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization/CMakeFiles/drl_localization_node.dir/depend.internal".
Scanning dependencies of target drl_localization_node
make[2]: 离开目录“/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization”
/usr/bin/make -f CMakeFiles/drl_localization_node.dir/build.make CMakeFiles/drl_localization_node.dir/build
make[2]: 进入目录“/home/zjh/cocobot-six-dynamic-2/build/dynamic_robot_localization”
[ 98%] Building CXX object CMakeFiles/drl_localization_node.dir/src/localization/localization_node.cpp.o
/usr/bin/c++  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"dynamic_robot_localization\" -Dqh_QHpointer -DvtkFiltersFlowPaths_AUTOINIT="1(vtkFiltersParallelFlowPaths)" -DvtkIOExodus_AUTOINIT="1(vtkIOParallelExodus)" -DvtkIOGeometry_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOImage_AUTOINIT="1(vtkIOMPIImage)" -DvtkIOParallel_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOSQL_AUTOINIT="2(vtkIOMySQL,vtkIOPostgreSQL)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL)" -DvtkRenderingCore_AUTOINIT="3(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingOpenGL)" -DvtkRenderingFreeType_AUTOINIT="2(vtkRenderingFreeTypeFontConfig,vtkRenderingMatplotlib)" -DvtkRenderingLIC_AUTOINIT="1(vtkRenderingParallelLIC)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL)" -I/home/zjh/cocobot-six-dynamic-2/devel/.private/dynamic_robot_localization/include -I/usr/include/vtk-6.3 -I/usr/include/freetype2 -I/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi -I/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent -I/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include -I/usr/lib/x86_64-linux-gnu/openmpi/include -I/usr/include/python2.7 -I/usr/include/x86_64-linux-gnu -I/usr/include/hdf5/openmpi -I/usr/include/libxml2 -I/usr/include/jsoncpp -I/usr/include/tcl -I/home/zjh/cocobot-six-dynamic-2/src/dynamic_robot_localization-1.2.0/include -I/usr/include/eigen3 -I/home/zjh/cocobot-six-dynamic-2/devel/include/pcl-1.8 -I/usr/include/ni -I/usr/include/openni2 -I/home/zjh/cocobot-six-dynamic/devel/.private/laserscan_to_pointcloud/include -I/home/zjh/cocobot-six-dynamic/src/laserscan_to_pointcloud/include -I/home/zjh/cocobot-six-dynamic/src/pose_to_tf_publisher/include -I/opt/ros/melodic/include -I/opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/usr/include/pcl-1.8 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++  -fopenmp    /home/zjh/cocobot-six-dynamic-2/devel/include/pcl-1.8 /usr/include/eigen3 /usr/include /usr/include/vtk-6.3 /usr/include/freetype2 /usr/lib/x86_64-linux-gnu/openmpi/include/openmpi /usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent /usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include /usr/lib/x86_64-linux-gnu/openmpi/include /usr/include/python2.7 /usr/include/x86_64-linux-gnu /usr/include/hdf5/openmpi /usr/include/libxml2 /usr/include/jsoncpp /usr/include/tcl /usr/include/ni /usr/include/openni2 -std=c++11 -fPIC -std=gnu++11 -o CMakeFiles/drl_localization_node.dir/src/localization/localization_node.cpp.o -c /home/zjh/cocobot-six-dynamic-2/src/dynamic_robot_localization-1.2.0/src/localization/localization_node.cpp
c++: warning: /home/zjh/cocobot-six-dynamic-2/devel/include/pcl-1.8: linker input file unused because linking not done
c++: warning: /usr/include/eigen3: linker input file unused because linking not done
c++: warning: /usr/include: linker input file unused because linking not done
c++: warning: /usr/include/vtk-6.3: linker input file unused because linking not done
[... cut ...]
[build]   Warnings:  None.                                                                                                                                                                        
[build]   Abandoned: None.                                                                                                                                                                        
[build]   Failed:    1 packages failed.                                                                                                                                                           
[build] Runtime: 1 minute and 4.8 seconds total. 

Dear God, my life is such a failure.

43998213, 2023-03-20:
> Hello,
> 
> In the log it shows includes from the /usr/include/pcl-1.8 installed from apt (this mixing of includes from /usr and my fork is causing this problem).
> 
> For compiling drl you need to compile [my fork of pcl](https://github.com/carlosmccosta/pcl) first. After compiling and sourcing the catkin setup.bash (this step is very important to ensure cmake uses the includes from my fork of pcl), when you do roscd pcl, it should go to the source folder of my pcl fork. Then, after compiling and sourcing the setup.bash, you can compile drl.
> 
> Check the install scripts:
> 
> * https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/install/e_build.bash
> * https://github.com/carlosmccosta/dynamic_robot_localization#installation-from-source
> 
> Also, checkout to the last commit of the default branches of drl and my fork of pcl and do a `catkin clean` before running my install scripts.
> 
> Check also other issues discussing this problem:
> 
> * [Localization failure | Generation of deb files #10 (comment)](https://github.com/carlosmccosta/dynamic_robot_localization/issues/10#issuecomment-538740122)
> * [build with ros-noetic #20](https://github.com/carlosmccosta/dynamic_robot_localization/issues/20)
> * [compile errors #30](https://github.com/carlosmccosta/dynamic_robot_localization/issues/30)
> 
> Have a nice day :)

I tried
zjh@ubuntu18-coco:/cocobot-six-dynamic-2$ roscd pcl
zjh@ubuntu18-coco:/cocobot-six-dynamic-2/src/pcl-pcl-1.8.1-cc-master-all-pr$ ls
you can see it in my log ,sorry is't too many.
thank you for your help，I will try again

qqngwzlb, 2023-03-23:
thanks

  

 
 
 
  
https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/install/e_build.bash
 
https://github.com/carlosmccosta/dynamic_robot_localization#installation-from-source