carlosmccosta/dynamic_robot_localization #36: 2D positioning error occurred, the positioning program selected the wrong pointcloud2 topic

chenhao2316, 2024-07-15:
Dear Mr. Carlos,

I tried using your outstanding work and its positioning effect was very excellent. But I have encountered a problem now, and I have tried to modify the code or configuration file, but none of them can solve it. I sincerely hope that you can point out the mistakes I have made.

The problem I encountered is that **/ambient_pointcloud** is getting farther and farther away from the wall. Then the positioning data also showed a few centimeters of error compared to the actual data.  **/ambient_pointcloud** becomes less attached to the wall as it runs, and then**/aligned_pointcloud** remains attached to the wall. I believe that the positioning data should be output by**/aligned_pointcloud** instead of**/ambient_pointcloud**, but the actual situation is that**/ambient_pointcloud** and radar**/scan** are attached, and the positioning data is also published based on**/ambient_pointcloud**.  

What is in red in the picture is **/aligned_pointcloud**，and what is in white in the picture is **/ambient_pointcloud**，It can be seen that **/aligned_pointcloud** is in contact with the wall.  And **/ambient_pointcloud** has error.

[image]

**/ambient_pointcloud** and **/scan** are matched:
[image]

Only when I give it another initial position value will everything match and become normal.
[image]

This is my TF data, one radar, one Odom:
[image]

Here is my configuration file:
[drl_configs.txt](https://github.com/user-attachments/files/16232501/drl_configs.txt)

launch file:
[launch.txt](https://github.com/user-attachments/files/16232533/launch.txt)

What can I do to make the location data based on**/aligned_pointcloud**?

Wishing you a wonderful day!

carlosmccosta, 2024-07-17:
Hello,

This is the expected behavior of drl.
By default, the `ambient_pointcloud` is on the `odom` tf frame, which accumulates error over time and the `aligned_pointcloud` is on the `map` tf frame, which is the corrected frame (aligned to map).

The pose published in topic `localization_pose` is based on the `aligned_pointcloud` and is the same as the tf chain `map` -> `base_link_frame_id` (configured on the launch file, which you seem to have set to `laser`).

Check in rviz with the global frame set to `map` and visualize the topic `localization_pose` as an arrow to track the pose estimated by `drl`.
Then move the robot to several map corners and check if the `drl` pose arrow is where it was expected to be.

Have a nice day,

chenhao2316, 2024-07-18:
Thank you for your reply. I have tried adjustments, I changed base_link_frame_id to base_link. 
 but the positioning data still seems to be based on ambient_pointcloud.

The laserscan and ambient_pointcloud fit together, and the localization_pose also coincides with the laser's TF, which is equivalent to saying that the localization_pose is based on the ambient_pointcloud instead of the aligned_pointcloud.
[image]

And only after I publish an initial value, the positioning data becomes correct.
before :
[image]
after(The two point clouds overlap and the positioning data becomes correct.):
[image]

I guess this may be related to TF, I am trying to solve it,

wish you have a good day.
[launch.txt](https://github.com/user-attachments/files/16280554/launch.txt)
[drl_configs.txt](https://github.com/user-attachments/files/16280555/drl_configs.txt)

carlosmccosta, 2024-07-18:
Hello,

It seems you have changed the [`publish_pose_tf_rate`](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L94) parameter to a positive value.

By default it has a negative value to indicate that the `map -> odom` tf will only be [published](https://github.com/carlosmccosta/pose_to_tf_publisher/blob/melodic-devel/src/pose_to_tf_publisher.cpp#L251) in the [TF chain](https://github.com/carlosmccosta/pose_to_tf_publisher/blob/melodic-devel/src/pose_to_tf_publisher.cpp#L275) after a [new pose](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L181) is estimated by drl (after a new laser scan alignment).

If you specify a positive value, the `map -> odom` will be published at a [fixed rate](https://github.com/carlosmccosta/pose_to_tf_publisher/blob/melodic-devel/src/pose_to_tf_publisher.cpp#L246), using the last alignment correction pose but with an updated time ([`ros::Time::now()`](https://github.com/carlosmccosta/pose_to_tf_publisher/blob/melodic-devel/src/pose_to_tf_publisher.cpp#L245)).

This functionality was added to allow other nodes to have available the tf `map -> base_link` between laser scan alignment corrections, allowing the usage of the odometry temporarily, until the next `map -> odom` correction is done from the next laser scan.

This might be useful if you have a motion controller running and constantly looking at the latest tf `map -> base_link` at 100 Hz, but have a laser scanning at 10 Hz, for example.
If you specify a positive value of 100 Hz for the `publish_pose_tf_rate`, the controller will get a valid tf `map -> base_link` at 100 Hz, but between each scan alignment (10 Hz), it will be relying on the odometry information to know where is the current pose of the robot on the map.

The drawback of this approach is that all the other nodes (including rviz), that look at tf to transform data to the map frame (which includes the laser scans), will have the error associated with odometry between scan alignment corrections.

Without this feature, the motion controller would block when calling the lookup transform api and would only be able to run at 10 Hz.

Alternatively, the motion controller could internally use the last `map -> odom` correction (10 Hz) and them combine it with the last odometry estimation (100 Hz). This is the best approach, that does not affect other nodes, such as rviz, and allows the motion controller to run at his own rate without being limited by the slow laser scanning.

That is why drl has by default the `publish_pose_tf_rate` parameter set to a negative value.

If you do not want this odometry error between scan alignment corrections, then you need to set the `publish_pose_tf_rate` to a negative value.

Also, the [drl_configs.yaml](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/yaml/schema/drl_configs.yaml) is only for documentation.

The yamls loaded to the parameter server and used by drl are the ones specified in the [dynamic_robot_localization_system.launch](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L153) along with the other roslaunch parameters inside the [node xml tag](https://github.com/carlosmccosta/dynamic_robot_localization/blob/noetic-devel/launch/dynamic_robot_localization_system.launch#L119).

Have a nice day,

chenhao2316, 2024-07-22:
Hello，
Thanks for your reply, 

I adjusted it as you said, and the drift speed dropped significantly, but when I ran it for more than half an hour, the same drift still occurred.

I suspect that there is something wrong with 
, because when I adjust the value of `<arg name="number_of_tf_queries_for_spherical_interpolation" default="4" />`
to 10, the location drifts a lot after running for two or three minutes, the actual position
 can be several meters away. 
[image]
[image]

Even if I keep the car stationary, it will still drift slowly.
[image]

So I tried to modify the laser scan assembler frame, I set it to laser, and now everything works fine.
```
	<!-- laser scan assembler -->
	<arg name="use_laser_assembler" default="true" />
	<arg name="assemble_lasers_on_map_frame" default="false" /> <!-- if false -> assembles lasers on odom frame -->
	<arg name="laser_assembly_frame" default="$(arg map_frame_id)" if="$(arg assemble_lasers_on_map_frame)" />
	<arg name="laser_assembly_frame" default="laser" unless="$(arg assemble_lasers_on_map_frame)" />
```

Have a good day.

carlosmccosta, 2024-07-22:
Hello,

By default the `laserscan_to_pointcloud` package uses the `odom` frame with 4 `number_of_tf_queries_for_spherical_interpolation` to correct for the scan deformation when the robot is moving.

But this feature will only work reliably if you have good odometry.
If there is still drift even with the robot stopped, it means you need to recalibrate your odometry.
With the robot stopped, the `odom` TF should stay the same and not change.

In the readme below you have more documentation and examples about the impact of the spherical linear interpolation correction in the final point cloud (generated from the laser scan).
https://github.com/carlosmccosta/laserscan_to_pointcloud

In the examples you can see that without the correction the first and last laser measurements do not overlap on the same wall line and on the left of the map the measurements do not overlap with the walls.

If you compare these images with the ones with the correction, you can see that these scan deformation issues are corrected.

Have a nice day,

chenhao2316, 2024-07-25:
Hello,
 I thought the drifting problem had been solved, but it seems not, it just appears much slower.

I have tried various methods in the past few days, such as disabling the laser assembler, replacing other point cloud merges and publishing **ambient_pointcloud** topic, or debugging the parameters in launch, or calibrating the odometer, but none of these worked, and the slow drift still occurred. It is worth mentioning that there is only a map to odom conversion in tf. 

The only way to solve this problem, I have found now, is to publish an init pose, and then trigger the tracking program, the icp registration. I am now trying to run it every thirty seconds.

I am going to modify the logic of your code. I think this is what should trigger the tracking, and this should solve the problem：
[image]

carlosmccosta, 2024-07-25:
Hello,

I found what was the problem.

This commit added the stop of the point cloud subscribers at the end of the drl pipeline when it exceeded the `limit_of_pointclouds_to_process` parameter without checking first if the limit was a number bigger than zero.
https://github.com/carlosmccosta/dynamic_robot_localization/commit/cac2ffcbf55b28a327a413f6a6da7ac7f39e8f5a#diff-7513768f17e2a1aa47eaba7a68a5e72984c98ca5ff3178706d67f52f24392e6bR2197

By default, `limit_of_pointclouds_to_process` is set to -1, which means, no limit is applied to the number of point clouds to process by drl.

I fixed this issue in last commit:
https://github.com/carlosmccosta/dynamic_robot_localization/commit/e7466e541e5784bd82c3ab31f0134c1c6ec8301b

Without this check for ensuring `limit_of_pointclouds_to_process` is bigger than zero, drl would process the first lidar scan and then stop processing any future sensor data.
In my precommit tests I had set a limit number higher than zero and did not took into account that the default value was -1.

That was why you were seeing the drift over time, because drl would process the first scan after you set the initial pose in rviz and then would stop the point cloud subscribers, which would result in drl stopping estimating the robot pose.

I apologize for the inconvenience and thank you very much for alerting me of this problem.

Please do `git stash` and `git pull` on the drl repository, compile and test again.

Have a nice day :)